BeamNG/BeamNGpy #125: No connection established between beamngpy and BeamNG.tech

Diminati, 2021-06-06:
Hello,
I am new here and I am trying to get the connection to the BeamNG.tech 0.22.3 and I trying to get the example in the Usage section of the documentation.
The Firewall setting in Windows 10 for domain and private are totally disabled. I run my script with pycharm (started as admin).
The BeamNG.tech interface opens properly but the connection between py and beamng doesn't seem to be established.

ERROR:root:[WinError 10061] No connection could be made because the target machine actively refused it
Traceback (most recent call last):
File "C:\Users\<username>\PycharmProjects\PythonTestProject\venv\lib\site-packages\beamngpy\beamng.py", line 472, in connect
    self.skt.connect((self.host, self.port))

I saw existing posts reporting a similar issue but the fix didn't really help to troubleshoot for me.
Am I missing anything for the initial setup?

pascale-commits, 2021-06-07:
Hi, 

excuse the convenience. Yes, since this BeamNG.tech release there is an initial setup step required before using the library. I've added the instructions to the readme on the dev branch. We'll release a new BeamNGpy version asap, but if that takes too long, you can check out and install the dev branch. I'll notify you as soon as the new version is released.

Diminati, 2021-06-07:
Hello Pascale, Thank you for the quick answer.
I finally managed to get it working with the current dev version.

I got the message "Cannot set up workspace if no licence key is available at <C:\BeamNG\BeamNG.tech.v0.22.3.0>"
I had to copy the license in that folder to be able to  setup the workspace properly. 
**_BeamNG.tech.key_** didn't work for some reason but worked once renamed as _**tech.key**_.

closed 2021-06-07 by Diminati

pascale-commits, 2021-06-08:
I've added the placement of the licence file to the description.