BeamNG/BeamNGpy #128: BeamNGpy 19.1 (and maybe 1.20) breaks compatibility with BeamNG 1.7.0.1

alessiogambi, 2021-06-24:
When running BeamNG.research 1.7.0.1 using beamngpy 1.19.1

Note that the same code works with BeamNG.tech 0.21.3.0.

Probably we should warn the user that BeamNG.research 1.7.0.1 is supported only up to 1.18?

The console log is attached below:

```
12.08113|D|GELua.core_modmanager.initDB|mountEntry -- /mods/beamngpy.zip:  : beamngpy
 12.10320|D|CEF.local://local/ui/lib/int/logger.js:70|MainGEUI | [Console] Expected AppLayout but didn't get one :-( 
 12.16793|D|GELua.core_modmanager.initDB|Notification : took 0.004ms to reorganise 3 files
 12.16811|D|GELua.core_modmanager.initDB|Notification : took 0.2138ms to callback
 12.27749|E|GELua.core_jobsystem.jobsystem|job error: [string "lua/ge/extensions//util/researchGE.lua"]:17: module 'jbeam/io' not found:
	no field package.preload['jbeam/io']
	no file 'lua/common/extensions//jbeam/io.lua'
	no file 'lua/ge/extensions//jbeam/io.lua'
	no file 'lua/ge/jbeam/io.lua'
	no file 'lua/gui/jbeam/io.lua'
	no file 'lua/common/jbeam/io.lua'
	no file 'lua/common/libs/jbeam/io/init.lua'
	no file 'lua/common/libs/luasocket/jbeam/io.lua'
	no file 'lua/jbeam/io.lua'
	no file 'jbeam/io.lua' / 
=============== Stack Traceback >> START >>
(1) Lua upvalue 'loadInternal' at line 348 of chunk 'lua/common/extensions.lua'
  Local variables:
  extName = string: "util_researchGE"
  extPath = string: "util/researchGE"
(2) Lua field 'load' at line 388 of chunk 'lua/common/extensions.lua'
  Local variables:
  moduleDataArray = table: 0x01fafff78e70  {1:ui_apps, 2:scenario_scenariosLoader, 3:campaign_campaignsLoader, 4:freeroam_freeroam (more...)}
  success = boolean: true
  extPath = string: "util/researchGE"
  (for generator) = C function: builtin#6
  (for state) = table: 0x01fafff78e70  {1:ui_apps, 2:scenario_scenariosLoader, 3:campaign_campaignsLoader, 4:freeroam_freeroam (more...)}
  (for control) = number: 43
  _ = number: 43
  extName = string: "util_researchGE"
(3) Lua global 'loadCoreExtensions' at file 'lua/ge/main.lua:209'
(4) Lua function '?' at line 561 of chunk 'lua/ge/extensions//core/modmanager.lua'
  Local variables:
  job = table: 0x01faffc5ff90  {t:thread: 0x01faffc5ffd8, args:table: 0x01faffc601c0, setExitCallback:function: 0x01faffc60358 (more...)}
  newMountedFiles = table: 0x01faffc605e0  {1:table: 0x01faddc01308, 2:table: 0x01fadddb71a0, 3:table: 0x01faddbabe20 (more...)}
  zipfileList = table: 0x01faffc60628  {1:/mods/BeamNGpy.zip}
  unpackedList = table: 0x01faffc606a8  {}
  fileList = table: 0x01faffc60738  {1:/mods/BeamNGpy.zip}
  mountList = table: 0x01faddb04168  {1:table: 0x01faddb04380}
  tim = 106.966s
  old_loadModule = Lua function '?' (defined at line 353 of chunk lua/common/extensions.lua)
  modScriptFiles = table: 0x01fafff78e28  {}
--------------- << END <<```

pascale-commits, 2021-06-29:
Hi, we support compatibility between minor BeamNGpy versions, so that users don't need to update their code when switching between minor BeamNGpy versions but we don't guarantee compatibility between any BeamNGpy version and any BeamNG.tech version. We'll add a table showing what versions match to the readme.

pascale-commits, 2021-08-04:
I've added a comment regarding this issue to the readme. ([Commit](https://github.com/BeamNG/BeamNGpy/commit/dde7c922307bc64669b073d8e66c703905b869a7))

closed 2021-08-04 by pascale-commits