BeamNG/BeamNGpy #129: Compatibility with steam BeamNG.drive

garethjns, 2021-07-06:
Hi,

I have the Steam version of BeamNG.drive (currently 0.23.1.0) and am wondering if I am able to use BeamNGpy with just this?

I see the readme mentions that the Steam version should be compatible, however I'm not able to follow the setup instructions. 

If I try and do the workspace set up, I hit an issue with it not finding a license file:
```
(beamng-env) >beamngpy setup-workspace C:\beamng_workspace_0.23
Could not set up workspace at <C:\beamng_workspace_0.23>.Note that this step is only neccessary for BeamNG.tech version 0.22 and above.
```

And if I continue with the example python code I get a connection error after calling `BeamNGpy.open()`. The game does launch, but it doesn't connect to the server.

```python
from beamngpy import BeamNGpy, Scenario, Vehicle

# Instantiate BeamNGpy instance running the simulator from the given path,
# communicating over localhost:64256
bng = BeamNGpy('localhost', 64256, home="S:/SteamLibrary/steamapps/common/BeamNG.drive", user="C:/beamng_workspace_0.23")
# Launch BeamNG.tech
bng.open()
```
```
No workspace set up at userpath: <C:\beamng_workspace_0.23>
Setup is required prior to mod deployment.
ERROR:root:Error connecting to BeamNG.tech. 25 tries left.
ERROR:root:[WinError 10061] No connection could be made because the target machine actively refused it
Traceback (most recent call last):
  File "C:\ProgramData\Anaconda3\envs\beamng-env\lib\site-packages\beamngpy\beamng.py", line 512, in connect
    self.skt.connect((self.host, self.port))
ConnectionRefusedError: [WinError 10061] No connection could be made because the target machine actively refused it

... [repeats for 25 tries]
```

Looking at the code, `BeamNGpy.read_effective_userpath(userpath)` always returns None as it can't find `researchHelper.txt`, presumably because the workspace isn't set up.

Am I missing something in the setup or is more required than just the Steam version to use BeamNGpy? 

Thank you.

pascale-commits, 2021-07-07:
Hi, 
yes, the user-path semantics changed in the latest BeamNG.tech/.drive version.
From now on you need to set up a workspace, as described in the [usage section](https://github.com/BeamNG/BeamNGpy#usage).

garethjns, 2021-07-07:
Hi @pascale-commits,

Thank you for the reply. I've tried to set up the workspace but it fails with the error above. 

I don't BeamNG.tech or a license file - are these required to set up the workspace in order to use BeamNGpy with the Steam version of BeamNG.drive?

pascale-commits, 2021-07-07:
Did you apply for a license yet? BeamNG.tech is free for academic use. You can find the registration form [here](https://register.beamng.tech/).

garethjns, 2021-07-07:
I haven't yet, I thought I'd try it with the Steam version first as the readme says in the prerequisites section:

> The regular Steam release of BeamNG.drive is compatible to an extent as well. Certain sensors like the simulated LiDAR or camera will not work, but most of the functions that are not exclusive to a Tech build will likely work.

But the .tech version is still required to use the Steam version?

pascale-commits, 2021-07-07:
I think its because this feature is also blocked on the game side by the missing license file. 
You can just manually create a `researchHelper.txt` and just write into it `C:\beamng_workspace_0.23\0.23\`.

garethjns, 2021-07-07:
That worked, thank you! It took a couple of extra steps:

1) I added an empty `researchHelpher.txt` to `c:\beamng_workspace_0.23\` (ie. not in the `0.23` subfolder) 
2) Then ran this script from `c:\a-different-path\readme_example.py`:
```python
from beamngpy import BeamNGpy, Scenario, Vehicle

from beam_al.config import Config

if __name__ == "__main__":
    # Instantiate BeamNGpy instance running the simulator from the given path,
    # communicating over localhost:64256
    bng = BeamNGpy('localhost', 64256, home="s:/SteamLibrary/steamapps/common/BeamNG.drive", user="C:/beamng_workspace_0.23")
    # Launch BeamNG.tech
    bng.open()
    # Create a scenario in west_coast_usa called 'example'
    scenario = Scenario('west_coast_usa', 'example')
    # Create an ETK800 with the licence plate 'PYTHON'
    vehicle = Vehicle('ego_vehicle', model='etk800', licence='PYTHON')
    # Add it to our scenario at this position and rotation
    scenario.add_vehicle(vehicle, pos=(-717, 101, 118), rot=None, rot_quat=(0, 0, 0.3826834, 0.9238795))
    # Place files defining our scenario for the simulator to read
    scenario.make(bng)

    # Load and start our scenario
    bng.load_scenario(scenario)
    bng.start_scenario()
    # Make the vehicle's AI span the map
    vehicle.ai_set_mode('span')
    input('Hit enter when done...')
```
3) This didn't work, but it did dump out `BeamNGpy.zip` to `c:\a-different-path\mods\beamNGpy.zip`. I copied this zip into `C:\beamng_workspace_0.23\0.23\mods` and reran the script, and the AI is now happily driving around.

I think the reason the `.zip` was written where it was is because `BeamNGpy.read_effective_userpath()` returns the contents of the empty file as `''`. Presumably that file usually contains a path?

Thanks again for your help.

pascale-commits, 2021-07-08:
yes, the researchhelper.txt is supposed to contain this string "c:\beamng_workspace_0.23\0.23\" and is supposed to be placed into the  <c:\beamng_workspace_0.23> Path :D

pascale-commits, 2021-07-16:
Can I close this issue?

closed 2021-07-16 by garethjns