BeamNG/BeamNGpy #132: Problems with newest release

DanAmador, 2021-07-11:
I decided to update to the v1.20 version as I needed a feature that was recently added. 

This in turn brought some problems:

The first one was that [this line in the scenario.py](https://github.com/BeamNG/BeamNGpy/blob/e0f2e648f044137a92f77f1e272dfcf851fc7ad9/src/beamngpy/scenario.py#L577) causes a KeyError. 
I assume the dictionary to check was cameras so I updated it in #131 

The second and more important issue is that vehicle.poll_sensors() is broken for me now. 
The console returns the  following exception whenever I call it:

```
022.947|E|GameEngineLua:Exception|[string "lua/ge/extensions//util/researchGE.lua"]:583: bad argument #3 to 'renderCameraShmem' (string expected, got boolean)
=============== Stack Traceback >> START >>
(1)  C function 'function: 0x015f20ddad10'
(2) field C function 'renderCameraShmem'
(3) Lua upvalue 'frameDelayFunc' at line 583 of chunk 'lua/ge/extensions//util/researchGE.lua'
  Local variables:
  otherData = nil
(4) Lua local 'funcListElement' at line 107 of chunk 'lua/ge/extensions//util/researchGE.lua'
  Local variables:
  dt = number: 0.0430973
(5) Lua field 'hook' at line 635 of chunk 'lua/common/extensions.lua'
  Local variables:
  func = string: "onPreRender"
  funcList = table: 0x015f20d80d78  {1:function: 0x015f21234ca8, 2:function: 0x015f5fff7300, 3:function: 0x015f2100c000 (more...)}
  (for index) = number: 1
  (for limit) = number: 10
  (for step) = number: 1
  i = number: 1
  funcListElement = Lua function '?' (defined at line 99 of chunk lua/ge/extensions//util/researchGE.lua)
  extCallName = nil
(6) Lua function 'luaPreRender' at file 'lua/ge/main.lua:299' (best guess)
  Local variables:
  dtReal = number: 0.0430973
  dtSim = number: 0.04175
  dtRaw = number: 0.0809019
  updateUIData = boolean: true
--------------- << END <<

```

Do you by any chance know what could be going on when trying to read the sensors. I'm only  using the cameras

pascale-commits, 2021-07-26:
Hi, please post your code so I can take a closer look at the issue :)

DanAmador, 2021-07-26:
I'm currently working on this project and didn't commit the exact code that caused the issue, but the closest commit I have to it [is this](https://github.com/DanAmador/CarCrasher/blob/6185c3a73e9018c0461d773411a338ea8fc5c34e/src/recorder.py#L95)  .

The vehicle reference was set and I followed the trace to [this method in BeamNGpy](https://github.com/BeamNG/BeamNGpy/blob/e0f2e648f044137a92f77f1e272dfcf851fc7ad9/src/beamngpy/vehicle.py#L316) where a response was never received. 

My code worked again after downgrading to version 1.19.

pascale-commits, 2021-08-04:
Hi, thanks for reporting but I've had no luck recreating this issue. I've tried to recreate your use-case but no luck. If you want to update BeamNGpy, you'll also need to use a compatible BeamNG.tech version. Maybe it is worth a try, but note that I haven't tried that myself. From the next update on we'll include a list of compatible BeamNG.tech - BeamNGpy versions.
I'll close this issue for now, but feel free to reopen it. At that time please include the python log file (use `setup_logging()`).

closed 2021-08-04 by pascale-commits