BeamNG/BeamNGpy #133: Accessing to BeamNG.tech from remote host

beunguk, 2021-07-16:
Hi, 

Currently, I have two machines that one is Windows10 and Ubuntu 18.04.
I installed BeamNG.tech v0.21.3 and BeamNGpy==0.19.1 on the windows machine, and I installed BeamNGpy==0.19.1 on Ubuntu.
If I understand correctly, BeamNG.tech supports multiple clients and I want to access BeamNG.tech from the Ubuntu machine.

So what I did is that I launched BeanNG.tech from the Windows machine. Then try to connect BeanNG.tech from the Ubuntu machine. On the BeamNG.tech console, it seems to accept the connection, but it closes or raises an error (timeout) after sending hello() or other msgs to BeamNG.tech.

So basically, it's like:

On windows, this is what I did, and it works. It launches BeamNG.tech with the console.
```python
from beamngpy import BeamNGpy, Scenario, Vehicle
# Instantiate BeamNGpy instance running the simulator from the given path,
# communicating over localhost:64256
bng = BeamNGpy('localhost', 64256)
# Launch BeamNG.tech
bng.open()
```

On Ubuntu, it works until bng.open(launch=False, deploy=False) , but at the hello() method, connection hangs.
And I modified some part of to skip checking binary file (.exe) existence in BeamNGpy package.
from beamngpy import BeamNGpy, Scenario, Vehicle
```python
# Instantiate BeamNGpy instance running the simulator from the given path,
# communicating over 'ip to the windows machine':64256
bng = BeamNGpy('ip to the windows machine', 64256)
# Launch BeamNG.tech
bng.open(launch=False, deploy=False) # it returns a BeamNGpy object as expected.
bng.hello()
```

When I am trying the above script, I got these logs from the console. (I also disabled all firewalls on windows and ubuntu.)
```
2806.073|I|GELua.util_researchGE.ResearchGE|Accepted new client: {Ubuntu Machine IP}/64256
2807.950|E|GELua.ResearchCom|Error reading from socket: timeout
2807.954|E|GELua.ResearchCom|Error reading from socket: tcp{client}: 0000020DCFB70500 - timeout
```

I know the setting is a bit complicated, but my development runs on a linux based system.
So I really want to access and control the simulator through the linux based system. Is there something I missed?

Palculator, 2021-07-19:
Hello! I tried a similar setup where I connected to BeamNG.tech from an Ubuntu VM after launching it on the Windows host system. If I run the code exactly as you pasted it, I can recreate the issue, but it seems to boil down to the basic fact that there is no code after the `bng.open` and `bng.hello` calls. Are the code examples you listed complete? If I add a simple `input("Press Enter when done...")` to the end of those snippets, the Ubuntu client stays connected to my Windows host.

On another note: The `bng.hello` call is unnecessary as it is done upon connection.

beunguk, 2021-07-20:
Maybe socket disconnection is not recognized until it really sends data.

When I run https://github.com/BeamNG/BeamNGpy/blob/master/examples/ai_line.py from this repository with some modification (changing IP, making launch option to False, etc). It still doesn't work, but on Windows machine it works well... Simply ticking like `while True: bng.step(60)` also cause the same issue.

Palculator, 2021-07-21:
Using a separate laptop running Linux and connecting to my Windows machine let's me replicate the issue. I don't have a fix or idea as to why it's happening yet, but at least I can reproduce it. Will let you know when I find out more.

Palculator, 2021-08-04:
Hey, sorry for the delay, but I pushed a fix to the `networkFix` branch that, at least for me, fixes the issue. I can run the `ai_line.py` example just fine with a similar setup you described using my own laptop running Linux and BeamNG.tech running on my Windows PC. Please clone that branch and use that version of the library to confirm the fix:
https://github.com/BeamNG/BeamNGpy/tree/networkFix

beunguk, 2021-08-09:
Thank you for the work, but when I tried with networkFix branch, but I am still having the same problem.

Following logs is what I get from Linux when I run `ai_line.py`.
```
[1] % python ai_line.py                                                                                                                                                       ~/workspace/BeamNG
2021-08-09 15:40:28,324 INFO     Started BeamNGpy logging.
2021-08-09 15:40:28,324 INFO     Opening BeamNGpy instance...
2021-08-09 15:40:28,324 INFO     Connecting to BeamNG.tech at: (172.20.0.129, 64256)
2021-08-09 15:40:28,325 INFO     Sent data!
2021-08-09 15:40:28,869 INFO     Connected!
2021-08-09 15:40:29,044 ERROR    Uncaught exception: 
Traceback (most recent call last):
  File "ai_line.py", line 76, in <module>
    main()
  File "ai_line.py", line 28, in main
    scenario.make(bng)
  File "/home/{user home}/.local/lib/python3.6/site-packages/beamngpy/scenario.py", line 733, in make
    self.path = bng.create_scenario(level_name, self.name, prefab, info)
  File "/home/{user home}/.local/lib/python3.6/site-packages/beamngpy/beamng.py", line 1959, in create_scenario
    level=level, name=name, prefab=prefab, info=info)
  File "/home/{user home}/.local/lib/python3.6/site-packages/beamngpy/beamng.py", line 373, in message
    resp = self.recv()
  File "/home/{user home}/.local/lib/python3.6/site-packages/beamngpy/beamng.py", line 553, in recv
    return recv_msg(self.skt)
  File "/home/{user home}/.local/lib/python3.6/site-packages/beamngpy/beamngcommon.py", line 173, in recv_msg
    length = skt.recv(16)
ConnectionResetError: [Errno 104] Connection reset by peer
2021-08-09 15:40:29,080 WARNING  sys:1: ResourceWarning: unclosed <socket.socket fd=3, family=AddressFamily.AF_INET, type=2049, proto=0, laddr=('172.20.0.28', 37686)>
```
or sometimes hang for a while after `2021-08-09 15:40:28,869 INFO     Connected!`, and got the same log output.
And I got this msg from Windows console.
```
001.592|I|BEAMNGCEF::CefManager::init|using CEF with software rasterizer
001.618|I|GELua.core_audio.onFirstUpdate|onFirstUpdate called....
003.465|E|BEAMNGCEF::BNGCefClient::OnResourceResponse|MainGEUI | local://local/ui/modules/mainmenu2/mainmenu2.js | response: 404 - ERROR
003.468|E|BEAMNGCEF::BNGCefClient::OnResourceResponse|MainGEUI | local://local/ui/modules/mainmenu2/mainmenu2.css | response: 404 - ERROR
003.542|E|BEAMNGCEF::BNGCefClient::OnResourceResponse|MainGEUI | local://local/ui/modules/mainmenu2/mainmenu2.js | response: 404 - ERROR
003.558|E|BEAMNGCEF::BNGCefClient::OnResourceResponse|MainGEUI | local://local/ui/modules/mainmenu2/mainmenu2.css | response: 404 - ERROR
010.597|I|GELua.util_researchGE.ResearchGE|Accepted new client: 127.0.0.1/64256
014.598|I|GELua.util_researchGE.ResearchGE|Accepted new client: 172.20.0.129/64256
014.625|E|GELua.ResearchCom|Error reading from socket: timeout
014.626|E|GELua.ResearchCom|Error reading from socket: tcp{client}: 00000230B8AD6510 - timeout
```
---

What I did is the following:
1. First, I launch BeanNG.tech with the following.
```python
def main():
    setup_logging()
    beamng = BeamNGpy('localhost', 64256)
    bng = beamng.open(launch=True)
    input("Press something")
if __name__ == '__main__':
    main()
```
2.  And run `ai_line.py` in Linux machine with following changes.
```python
   15 def main():
   16     setup_logging()
   17 
   18     beamng = BeamNGpy('172.20.0.129', 64256, remote=True)
   19     bng = beamng.open(launch=False, deploy=False)
   20 
   21     scenario = Scenario('west_coast_usa', 'ai_sine')
...
```

Thanks.

closed 2026-01-05 by EPI-BeamNG