BeamNG/BeamNGpy #135: What's the maxium reading frequency of imu sensor

hyf3513OneGO, 2021-07-18:
I was tring  to read imu sensors at 50ms interval and it did work for a moment but suddenly,the python console shows a fatal error about encoding not right
[image]

In oreder to fix i,I put an output code(_print(length)_) in beamngpy.beamngcommon.py 
tring to get the message the socket received
[image]

Actually it did show something unusual before crash
[image]

I guess,this may be caused by socket-jam
here is my code(only part with error output) and traceback
[code](https://paste.ubuntu.com/p/bbMvyPrkM2/)
[trace_back](https://paste.ubuntu.com/p/WJ2DxBqs38/)

----------------------------------------------------------------------------------------------------------------------------------------------
os: windows 10 20H1
python_version: python3.9(pipenv)
language:zh-CN

pascale-commits, 2021-07-26:
Hi, 
why are you using threads? 
I recommend to iteratively run the simulation a number of steps and then execute `poll_sensors`. If you make a deep copy of the data that is then processed in a different thread it might work.
Do not use `poll_sensors` to access data, but only if you need to update data. Use `v1.sensors['imu_chest_M'].data['aX']` for data access instead.
Know that the IMU has problems right now, we will try to fix it asap.
See #121 for reference.

In the future, please do not use screen shots to share code, console output, etc. Only use them if there are problems on the graphics side and try to create a minimal code example to recreate the bug.

closed 2021-07-27 by hyf3513OneGO

Palculator, 2021-08-05:
To answer the actual question: The reading rate is not bound by Python. Communication between BeamNG.tech and BeamNGpy is tied to the frame rate of the simulator, including any IMU measurements. Your maximum reading rate therefore becomes your FPS. Besides this general limitation, IMUs also only update their values once per frame. There is a higher rate IMU system that exists only for the refNode of a vehicle (so it can't be placed at arbitrary points) and would have data available at 2000Hz, but that rate would still be capped by the rate of message exchange.