BeamNG/BeamNGpy #149: Version 1.18 cannot properly installed from PyPI with `pip`

ChristianBirchler, 2021-11-17:
**User story:** 
As a developer relying on `beamngpy v1.18` I want to provide a dependency list in my repository so that the user of my project can install all the specific dependencies with `pip`.

**Issue:**
The `beamngpy v1.18` can not be downloaded properly from [PyPI](pypi.com). It seems there is an issue with the metadata. Below you find the output of my shell trying to install `beamngpy v1.18` with `pip`.

```
Collecting beamngpy==1.18
  Downloading beamngpy-1.18.tar.gz (24.3 MB)
     |████████████████████████████████| 24.3 MB 5.5 MB/s
  WARNING: Requested beamngpy==1.18 from https://files.pythonhosted.org/packages/40/88/ab8a7760730fcad1fc9b61e0a78f006ebd5c33a39495ddc12a1add749507/beamngpy-1.18.tar.gz#sha256=efe863f00576e8fd82f350b5124daa7a534f873ba89d667e1b980232cc2ac0e2, but installing version 0.0.0
WARNING: Discarding https://files.pythonhosted.org/packages/40/88/ab8a7760730fcad1fc9b61e0a78f006ebd5c33a39495ddc12a1add749507/beamngpy-1.18.tar.gz#sha256=efe863f00576e8fd82f350b5124daa7a534f873ba89d667e1b980232cc2ac0e2 (from https://pypi.org/simple/beamngpy/). Requested beamngpy==1.18 from https://files.pythonhosted.org/packages/40/88/ab8a7760730fcad1fc9b61e0a78f006ebd5c33a39495ddc12a1add749507/beamngpy-1.18.tar.gz#sha256=efe863f00576e8fd82f350b5124daa7a534f873ba89d667e1b980232cc2ac0e2 has inconsistent version: filename has '1.18', but metadata has '0.0.0'
ERROR: Could not find a version that satisfies the requirement beamngpy==1.18 (from versions: 0.1.1, 0.1.2, 0.1.3, 0.2, 0.3, 0.3.1, 0.3.2, 0.3.3, 0.3.4, 0.3.5, 0.3.6, 0.4, 0.19.0, 1.0, 1.1, 1.3, 1.4, 1.5, 1.6, 1.7, 1.7.1, 1.8, 1.9, 1.9.1, 1.10, 1.11, 1.12, 1.13, 1.14, 1.14.1, 1.15, 1.16, 1.16.1, 1.16.2, 1.16.3, 1.16.4, 1.16.5, 1.17, 1.17.1, 1.18, 1.19.0, 1.19.1, 1.20, 1.21)
ERROR: No matching distribution found for beamngpy==1.18
WARNING: You are using pip version 21.1.1; however, version 21.3.1 is available.
You should consider upgrading via the '/Users/birc/temp/venv/bin/python -m pip install --upgrade pip' command.
(venv) ➜  temp
```

aivora-beamng, 2021-11-19:
Hi, it should be fixed now. Try to install BeamNGpy again with:
`pip install --no-cache-dir beamngpy==1.18`

ChristianBirchler, 2021-11-19:
Now it works. Thanks a lot.

closed 2021-11-19 by ChristianBirchler