BeamNG/BeamNGpy #150: scenario.make isn't working

RoboticistAbdulrahman, 2021-11-24:
PS D:\BeamNg.Tech_ws\BeamNG.drive-0.23.5.1.12888> python

Python 3.9.9 (tags/v3.9.9:ccb0e6a, Nov 15 2021, 18:08:50) [MSC v.1929 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from beamngpy import BeamNGpy, Scenario, Vehicle
>>> bng = BeamNGpy('localhost', 64256, home='D:\BeamNg.Tech_ws\BeamNG.drive-0.23.5.1.12888') 
>>> scenario = Scenario('west_coast_usa', 'example')
>>> vehicle = Vehicle('ego_vehicle', model='etk800', licence='PYTHON')
>>> scenario.add_vehicle(vehicle, pos=(-717, 101, 118), rot=None, rot_quat=(0, 0, 0.3826834, 0.9238795))
>>> scenario.make(bng)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\ideas\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\beamngpy\scenario.py", line 789, in make   
    self.path = bng.create_scenario(level_name, self.name, prefab, info)
  File "C:\Users\ideas\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\PythonythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\beamngpy\beamng.py", line 375, in message
    self.send(kwargs)  File "C:\Users\ideas\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\beamngpy\beamng.py", line 548, in send
    return send_msg(self.skt, data)  File "C:\Users\ideas\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\beamngpy\beamngcommon.py", line 238, in send_msg    skt.send(bytes(length, 'ascii'))AttributeError: 'NoneType' object has no attribute 'send'

aivora-beamng, 2021-11-26:
Hi, it seems that you are missing a `bng.open()` call in your code. The game has to be open for the scenario to be made.

With the following snippet, I do not encounter the problem you are describing:
```python
bng = BeamNGpy('localhost', 64256, home='D:\BeamNg.Tech_ws\BeamNG.drive-0.23.5.1.12888')
scenario = Scenario('west_coast_usa', 'example')
vehicle = Vehicle('ego_vehicle', model='etk800', licence='PYTHON')
scenario.add_vehicle(vehicle, pos=(-717, 101, 118), rot=None, rot_quat=(0, 0, 0.3826834, 0.9238795))
bng.open()
scenario.make(bng)

bng.load_scenario(scenario)
bng.start_scenario()
```

RoboticistAbdulrahman, 2021-11-26:
i usually can't load any scenario either i used bng.open() or not 

`PS D:\BeamNg.Tech_ws\BeamNG.drive-0.23.5.1.12888> python.exe 
Python 3.9.9 (tags/v3.9.9:ccb0e6a, Nov 15 2021, 18:08:50) [MSC v.1929 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from beamngpy import BeamNGpy, Scenario, Vehicle
>>>     bng = BeamNGpy('localhost', 64256, home='D:\BeamNg.Tech_ws\BeamNG.drive-0.23.5.1.12888') 
  File "<stdin>", line 1
    bng = BeamNGpy('localhost', 64256, home='D:\BeamNg.Tech_ws\BeamNG.drive-0.23.5.1.12888') 
>>> bng.open()
No workspace set up at userpath: <C:\Users\ideas\AppData\Local\BeamNG.drive>. Setup is required prior to mod deployment.
Error connecting to BeamNG.tech. 25 tries left.
[WinError 10061] No connection could be made because the target machine actively refused it
Traceback (most recent call last):
  File "C:\Users\ideas\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\beamngpy\beamng.py", line 527, in connect
    self.skt.connect((self.host, self.port))
ConnectionRefusedError: [WinError 10061] No connection could be made because the target machine actively refused it
Error connecting to BeamNG.tech. 24 tries left.
[WinError 10061] No connection could be made because the target machine actively refused it
Traceback (most recent call last):
  File "C:\Users\ideas\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\beamngpy\beamng.py", line 527, in connect
    self.skt.connect((self.host, self.port))
ConnectionRefusedError: [WinError 10061] No connection could be made because the target machine actively refused it
Error connecting to BeamNG.tech. 23 tries left.
[WinError 10061] No connection could be made because the target machine actively refused it
Traceback (most recent call last):`
  File "C:\Users\ideas\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\beamngpy\beamng.py", line 527, in connect
    self.skt.connect((self.host, self.port))
    
    
    

[image]

RoboticistAbdulrahman, 2021-11-26:
thanks adam, i had a wrong setup in my license, thanks for helping me through it

closed 2021-11-26 by RoboticistAbdulrahman