BeamNG/BeamNGpy #157: support for beamng.drive

JustAnotherGameDevWasTaken, 2022-01-12:
i already have beamng.drive and i want to use beamng,py with it, but it doesn't connect, does beamng.py still support beamng.drive, or do i need beamng.tech / a license for beamng.tech? it launches the game, but doesn't connect, and i can't configure a workspace. i have the firewall allowing traffic through the port it uses, but it still doesn't connect.

aivora-beamng, 2022-01-17:
Hi, although it is recommended to use BeamNGpy with the BeamNG.tech product, you can use some features of the BeamNGpy library with BeamNG.drive (without configuring the workspace, you need to have a tech license key for that). I have tested the latest version of BeamNGpy with the latest Steam version of BeamNG.drive and the "Object placement" demo notebook and it works as expected.

Try to upgrade your BeamNGpy to the latest version using:
```bash
pip install --upgrade beamngpy
```

Alternatively, you can apply for a BeamNG.tech license at https://register.beamng.tech/.

JustAnotherGameDevWasTaken, 2022-01-17:
it stops complaining about the license if i create a blank tech.key file, but then it complains about not having a valid home path, despite me setting one in the environment variables, and it still refuses to connect to the game

aivora-beamng, 2022-01-18:
Ah, I see the problem now. The setup-workspace command will not work without a valid tech.key, but it should be possible to setup the workspace manually. Try these steps:
1. Backup and delete the BeamNG user folder (default location is `%localappdata%\BeamNG.drive`).
2. Run BeamNG.drive from Steam to regenerate the user folder and then close the game.
3. In your user folder (`%localappdata%\BeamNG.drive`), create a new text file `researchHelper.txt` containing a single line: `C:\Users\<USERNAME>\AppData\Local\BeamNG.drive\0.24\`, where you replace `<USERNAME>` with your Windows username.
 4. You should now be able to use BeamNGpy with BeamNG.drive:
```python
from beamngpy import BeamNGpy, Scenario, Vehicle
bng = BeamNGpy('localhost', 64256, home=r'C:\Program Files (x86)\Steam\steamapps\common\BeamNG.drive')
bng.open()
scenario = Scenario('west_coast_usa', 'example')
vehicle = Vehicle('ego_vehicle', model='etk800', licence='PYTHON')
scenario.add_vehicle(vehicle, pos=(-717, 101, 118), rot=None, rot_quat=(0, 0, 0.3826834, 0.9238795))
scenario.make(bng)
bng.load_scenario(scenario)
bng.start_scenario()
vehicle.ai_set_mode('span')
```

aivora-beamng, 2022-02-07:
Closing this, hopefully the steps mentioned before should fix your problem. If it still persists, please create another issue.

closed 2022-02-07 by aivora-beamng