BeamNG/BeamNGpy #160: Spawning vehicles at speed (accel, jerk) greater than zero

alessiogambi, 2022-02-15:
It should be possible to spawn new vehicles at a given (physical) state, i.e., with a given speed, acceleration, etc..

Diminati, 2022-03-14:
Totally agree! I would like to see this feature being added, especially on the ai_set_script for example.

Some heavy vehicles take a lot of time to get up to speed and that's a lot of time lost for each run.
That would also help the procedural creation of scenarios with multiple cars.

alessiogambi, 2022-03-29:
@aivora-beamng @dstark481 Is there any progress or news about this feature?

aivora-beamng, 2022-03-29:
Hi, this feature should make it into the next major release (BeamNG.tech 0.25).

aivora-beamng, 2022-06-16:
Implemented in BeamNG.tech 0.25 / [BeamNGpy 1.23](https://github.com/BeamNG/BeamNGpy/tree/v1.23).

On a spawned vehicle, you can use the [set_velocity](https://beamngpy.readthedocs.io/en/latest/beamngpy.html#beamngpy.vehicle.Vehicle.set_velocity) method.

closed 2022-06-16 by aivora-beamng