BeamNG/BeamNGpy #162: How to work with Lidar

ASionville, 2022-02-26:
Hey!
I'm currently trying to figure out how does the lidar system works.
I would like to figure out which point (so which direction) is the nearest "obstacle" in front of the car.

I have set up the Lidar as default, I only disabled the visualization.
When I get `vehicle.sensors["lidar"].data["points"]`, it gives me a whole lot of floats numbers that don't seem to meen anything.
Many of those are negative values, from -1000 to 0.
If they represent a distance, why are they negative, and even though, how come they are -900? Isn't the lidar max range 200 by default?

What does these values represent ?
Do you have a clue how I can get the point supposed to be the closest one? I'll do the 3D math by myself afterward.

dstark481, 2022-02-28:
Hi Asionville,

Thanks for the question.

The LIDAR sensor returns a collection of 3D points, which represent all the world-space positions of obstacles which it can "see" in all directions within its range.  It returns an array which is formatted as a linear sequence of coordinate triplets in the form of [x0, y0, z0, x1, y1, z1, ..., xn, yn, zn].   So the reason you are seeing negative values is because this is actually one of the world-space dimensions of the point, rather than an actual distance.  To convert these values to distances from the LIDAR sensor itself, one would need to compute the distance between each point in this array and the position of the sensor, eg using the 3D distance formula Sqrt((x - x0)^2 + (y - y0)^2  + (z - z0)^2 ). 

I hope this helps.  Let me know how you get on, and if there are still problems we can take a look.

Dave Stark,
Senior Research Software Engineer,
BeamNG

ASionville, 2022-02-28:
Hi !
Thanks for the information, helped me a lot to figure out what's going on.

I did those functions if it can help anyone out there :
```python
def compute_spherical_coordinates(pointA:tuple, pointB:tuple):
    """
    Compute the spherical coordinates of a point from its cartesian coordinates.
    """
    # Get the cartesian coordinates of the point
    x = pointB[0] - pointA[0]
    y = pointB[1] - pointA[1]
    z = pointB[2] - pointA[2]
    # Compute the spherical coordinates
    r = (x**2 + y**2 + z**2)**0.5
    theta = math.acos(z/r)
    phi = math.atan2(y, x)
    return r, theta, phi

def compute_cartesian_coordinates(r:float, theta:float, phi:float, origin:tuple):
    """
    Compute the cartesian coordinates of a point from its spherical coordinates.
    """
    x = r*math.sin(theta)*math.cos(phi) + origin[0]
    y = r*math.sin(theta)*math.sin(phi) + origin[1]
    z = r*math.cos(theta) + origin[2]
    return x, y, z

def get_closest_point(lidar_points:list, origin:tuple):
    """
    Get theta and phi angle corresponding to the closest point to the origin in the list of lidar points.
    Args:
        lidar_points (list): List of points in the lidar.
        origin (tuple): Origin of the lidar.
    Returns:
        min_distance (float): Minimum distance between the origin and the closest point.
        theta (float): Theta angle of the closest point.
        phi (float): Phi angle of the closest point.
    """
    spherical_coords = [compute_spherical_coordinates(origin, point) for point in lidar_points]
    min_distance, theta, phi = min(spherical_coords, key=lambda x: x[0])
    return min_distance, theta, phi

def list_of_coordinates_to_points(coordinates:list):
    """
    Convert a list of coordinates to a list of points.
    Args:
        coordinates (list): List of coordinates [x1, y1, z1, x2, y2, z2, ...].
    Returns:
        points (list): List of points.
    """
    points = []
    for i in range(0, len(coordinates), 3):
        points.append((coordinates[i], coordinates[i+1], coordinates[i+2]))
    return points

def lidar_to_closest_point(lidar_points_coords:list, origin:tuple):
    """
    Convert a list of lidar point coordinates to the closest point to the origin.
    Args:
        lidar_points_coords (list): List of coordinates of the lidar points [x1, y1, z1, x2, y2, z2, ...].
    Returns:
        point (tuple): Distance, theta and phi angles of the closest point.
    """
    return get_closest_point(list_of_coordinates_to_points(lidar_points_coords), origin)
```

closed 2022-02-28 by ASionville

NotInControl, 2025-02-08:
Hi thank you this conversation is extremely helpful.

Is this information of what a pointcloud is defined as and what is returned by thr lidar system in the documentation? I struggled to find it. If not can we contribute to the documentation?

Also is there a way to find the orientation of the lidar with respect to the world.

I assume we can use lidar.get_direction() ans lidar.get_position() to form a 3x3 rotation matrix but before I do that on my own wanted to know if that's needed?

Thanks for this by the way it's awesome

dstark481, 2025-02-08:
Not sure I understand the question about the point cloud.  Depending on which mode you use the LiDAR in, you will have a collection of returns in the form of world-space points.  In the default mode, these should span the full 360 degrees range.  We can't simulate rotation very well when the LiDAR rotation frequency is faster than the frame rate of the simulator, so we return the full circle of hits.  You can think of the point cloud as 'raw data', which can be processed further to provide something more meaningful to your particular application (eg closest point).

It is possible to get the orientation: In BeamNGPy, the 'up' vector of the LiDAR frame is not exposed unfortunately, but since you set the sensor relative to the vehicle, you could get the local frame from the vehicle using the state sensor, where there are 'dir' (the unit forward vector) and 'up'.  There is also a quaternion in there called 'rotation' (I am just looking at the code comments here), if you prefer to use that.  The same is true for position - you set the LiDAR position relative to the car, so you can use the current vehicle position for the state sensor ('pos') then include the relative position of the sensor (done in the vehicle's reference frame), using some vector work with the forward, up, and right vectors used as a basis.

If that's too much, you could just assume the 'up' vector is always (0, 0, 1).  This of course will only be accurate for when the vehicle is driving on flat (or near-flat) ground, but this is a typical case.

Best,
Dave