BeamNG/BeamNGpy #165: BeamNGpy not working with Flowgraph Scenarios

treins, 2022-04-14:
I can't make BeamNGpy work with Flowgraph scenarios. 

I created a simple Scenario with Flowgraph. It doesnt have a start screen or a countdown. The Problem is, bng.load_scenario() never finishes. If I add a start screen, bng.load_scenario() works as it should. But bng.start_scenario() isn't working and I have to start the scenario manually.

Any ideas?

stevenlee090, 2022-07-13:
Hi @treins, did you figure this out? I am also curious if flowgraph can work with beamngpy.

aivora-beamng, 2022-07-26:
Hi,
Flowgraph scenarios are unfortunately not supported at this moment. We will be adding support in the next BeamNG.tech release.

aivora-beamng, 2022-09-20:
Hello,
we released the support for Flowgraph scenarios in BeamNGpy 1.24 and BeamNG.tech 0.26. You need to update both the simulator and the Python library. Then you can try this snippet to load the 'Gravity Fuel' scenario, which is Flowgraph-based. The code should also work with your custom Flowgraph scenario. If not, please let us know.

```python
from beamngpy import BeamNGpy, Scenario
import time

beamng = BeamNGpy('localhost', 64256)
beamng.open()

scenario = Scenario('italy', 'Gravity Fuel', path=r'/levels/italy/scenarios/noFuel_v2/gravityFuel.json')
beamng.load_scenario(scenario)
vehicles = {vehicle.model: vehicle for vehicle in scenario.vehicles}
autobello = vehicles['autobello']

beamng.start_scenario()

time.sleep(5)
for i in range(10):
    if i > 5:
        autobello.control(throttle=1.0, steering=1.0)
    autobello.poll_sensors()
    print(autobello.sensors['state'].data['pos'])
    
    time.sleep(1)
```

aivora-beamng, 2022-10-05:
Closing this, if you have any issues with Flowgraph scenarios, feel free to reopen.

closed 2022-10-05 by aivora-beamng