BeamNG/BeamNGpy #166: API not working with BeamNG.drive

quaidr, 2022-04-14:
The documentation mentions that BeamNGpy should have basic support for BeamNG.drive:

> The regular Steam release of BeamNG.drive is compatible to an extent as well. Certain sensors like the simulated LiDAR or camera will not work, but most of the functions that are not exclusive to a Tech build will likely work.

But I cannot get it to work. It seems that it fails to load the util_researchGE extension and with that the game will not respond on the API port 64256. I tried to follow earlier reported (now closed) issues on how to manually set up a workspace, but it does not seem to work for me. Is that the reason why the util_researchGE extension does not work? What am I missing?

BeamNG.drive console output:
000.623|I|Paths|User path: C:\Users\quaid\AppData\Local\BeamNG.drive\0.24\
002.941|I|GELua.core_input_bindings.bindings|Controller connected: XBox Controller 1 (xinput0/0xXIDevice)
002.977|I|GELua.print|*************LightRayPostFX 2 set
002.982|I|GELua.print|********PostFXSSAOGeneralEnabled 2 set
003.033|I|reloadLanguages|Switched from '' language to 'en-US' (as fallback, after trying 'sv-SE')
003.382|W|GELua.core_levels.|info.json missing: /levels/GridMap/info.json
003.383|E|GELua.core_levels.|No entry point for level found: /levels/GridMap. Ignoring level.
003.733|E|GELua.extensions|extension unavailable: "util_researchGE" at location: "util/researchGE"

Python output:
python .\west_coast_random.py
2022-04-15 01:20:39,475 |INFO     |beamngpy                      |Started BeamNGpy logging.
2022-04-15 01:20:39,475 |WARNING  |py.warnings                   |C:\Users\quaid\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\beamngpy\beamng.py:69: DeprecationWarning: The use of `beamng.setup_logging` is deprecated and will be removed in future versions. Use `beamngcommon.set_up_simple_logging` or `beamngcommon.config_logging` instead.
  create_warning(warn_msg, DeprecationWarning)

2022-04-15 01:20:39,475 |WARNING  |py.warnings                   |C:\Users\quaid\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\beamngpy\beamng.py:69: DeprecationWarning: The use of `beamng.setup_logging` is deprecated and will be removed in future versions. Use `beamngcommon.set_up_simple_logging` or `beamngcommon.config_logging` instead.
  create_warning(warn_msg, DeprecationWarning)

2022-04-15 01:20:39,681 |INFO     |beamngpy.BeamNGpy             |Opening BeamNGpy instance.
2022-04-15 01:20:39,683 |INFO     |beamngpy.BeamNGpy             |Started BeamNG.
2022-04-15 01:20:42,006 |INFO     |beamngpy.BeamNGpy             |Terminating BeamNG.tech process.
2022-04-15 01:20:42,051 |INFO     |beamngpy.BeamNGpy             |Started BeamNG.
2022-04-15 01:20:44,323 |INFO     |beamngpy.BeamNGpy             |Terminating BeamNG.tech process.
2022-04-15 01:20:44,362 |ERROR    |beamngpy.beamng               |No workspace set up at userpath: <C:\Users\quaid\AppData\Local\BeamNG.drive>. Setup is required prior to mod deployment.
2022-04-15 01:20:44,365 |INFO     |beamngpy.BeamNGpy             |Started BeamNG.
2022-04-15 01:20:54,367 |INFO     |beamngpy.BeamNGpy             |Connecting to BeamNG.tech at: (localhost, 64256)
2022-04-15 01:20:56,390 |ERROR    |beamngpy.BeamNGpy             |Error connecting to BeamNG.tech. 25 tries left.
...

aivora-beamng, 2022-04-19:
Hi,
I followed the steps in https://github.com/BeamNG/BeamNGpy/issues/157#issuecomment-1015330607 for the latest Steam version of the game (0.24.1.3) and the latest version of BeamNGpy (1.22).

It seems that the mod deployment part did not work, in case of a successful mod deployment, there should exist a `BeamNGpy.zip` file in the `C:\Users\<USERNAME>\AppData\Local\BeamNG.drive\0.24\mods\` directory. Does it exist in your case?

Please try to double-check that you are doing the same steps as I did.

quaidr, 2022-04-20:
Hi,
I checked and indeed I do not have the BeamNGpy.zip file in the ..\0.24\mods\ directory. How do I get this file? I tried to follow the steps in the issue you linked to, to regenerate the BeamNG user folder, but that does not add the BeamNGpy.zip file to the folder? Which step have I missed?

Thanks for your help!

aivora-beamng, 2022-04-21:
Have you also created the `researchHelper.txt` file? If so, `BeamNGpy.zip` should be generated with the first call to the `open` method of the `BeamNGpy` object:
```python
bng = BeamNGpy('localhost', 64256, home=r'C:\Program Files (x86)\Steam\steamapps\common\BeamNG.drive')
bng.open() # here, the BeamNGpy.zip should be automatically created in the mods folder
```

If that doesn't happen, as a workaround you can try to use this already generated [BeamNGpy.zip](https://github.com/BeamNG/BeamNGpy/files/8529493/BeamNGpy.zip), move it manually into the `..\0.24\mods` folder, and try to run your `BeamNGpy` code again.

Be aware that the features of BeamNGpy are quite limited for BeamNG.drive. If you are an academic/non-commercial user and have an interest in BeamNG.tech, you can try to apply for a BeamNG.tech license at https://register.beamng.tech/.

quaidr, 2022-04-21:
Thanks, it works now with the zip-file you provided.

closed 2022-04-21 by quaidr