BeamNG/BeamNGpy #169: LIDAR access error

mayureshb7, 2022-06-06:
Hi

Whenever I try to open BeamNG and a scenario it does not have any problem and when I try to run LIDAR visualization or the example code lidar_point_cloud_data.py it opens BeamNG but throws following error. Can someone provide a solution to it.

[image]

dstark481, 2022-06-08:
Hi Mayuresh,

Thanks for your email. It appears to be a mix-up on our end.  You are using a future version of BeamNGPy that has been pushed to the github repo a little too early.  Our release is scheduled for next week, so we should have waited to push this until that time.  The sensors in this version of beamngpy (in master) is not compatible with the 0.24 version of BeamNG.tech.

We have reverted the github master back, so please now reclone the repo and try again with the existing python scripts which are there.  That specific script you reference 'lidar_point_cloud_data.py' is only relevant for the upcoming version, so it will not be available until then.  We apologise for the mistake.

Best regards,
BeamNG Tech team.  

https://github.com/BeamNG/BeamNGpy/issues/169

mayureshb7, 2022-06-09:
Hi Team,

Thank you for your response. I have cloned the latest version of BeamNGPY, and I can access LIDAR data now.
However, I would like to understand how the data is represented in terms of World coordinates and Ego vehicle coordinates. 
I'm trying to create an obstacle avoidance course by object detection in an off-road environment map or BeamNG. Do you have any suggestions that I should be working on to successfully implement the algorithm for the same.

Also as per below mentioned image, when I print self.vehicle.state data I get direction, rotation, position, front, up and velocity of the vehicle. I would like to know what dir, rotation, front and up represents.

[image]

Warm Regards,
Mayuresh Bhosale

dstark481, 2022-06-09:
Hi Mayuresh,

Thanks for your query.  We are glad our LiDAR sensor is working for you now.  

In order to help answer your first query, there are two spaces which must be considered: 
1. The 'world space', which contains all of the map geometry. This is a standard (x, y, z) space in metres, with z being elevation. This space has an origin point (0, 0, 0) somewhere on the map. The LiDAR point data which you are able to see exists in this space.
2. There is also 'vehicle space'. This is similar to the world space, but is used as a reference space for the vehicle to which it relates. This means the origin will be in the vehicle rather than at a fixed point on the map. This is a more useful space to describe any vehicle-related properties, since it is local to a specific vehicle. 

The only difference between this space and world space is in its linear translation; that is to say that the rotation of any vector in this space is the same as it is in the world space.  
To go from vehicle space to world space, for example, we can fetch the vehicle position and add this to any vehicle point, and we will get the world space position of that vehicle point.  One can also go from world space to vehicle space by subtracting instead. Note that we do not need a transformation matrix to go between the two spaces - a vector is sufficient.

Regarding the vehicle state properties:
'dir' is a 3D vector which describes the direction of the vehicle (in which direction the vehicle faces forward).
'up' is a generalised 'upwards' vector for the vehicle, typically you could imagine an arrow pointing straight up from the vehicle centre. Note however that this is not necessarily orthogonal (at right angles) to the forward direction, and can change in relation to the forward direction if the vehicle becomes deformed.
'rotation' is actually a quaternion (hence why it contains 4 values). This describes the rotation of the vehicle in 3D space. This could be converted to a matrix or Euler angles, if required. We use quaternions internally for performance and efficiency reasons.
'front' is the position coordinate of the front of the vehicle in world space.

Object detection can be very complicated problem.  We ourselves are not experts in this area, although we do offer useful simulation tools to help people who research it. Like most simulation problems, there are simple, heuristic approaches and complicated ones (usually based on mathematical theory) and/or involving AI and sensor fusion techniques. Research in this area is still very active, from what I can see. 
I can try and offer some basic suggestions for you to get started, but this is simply what I myself might try if I wanted to solve a simple object detection problem.

1. Firstly, the LiDAR is a good choice of sensor for obstacle detection because it gives you not only the distance of objects but also their direction. You can work this out by computing the vectors from the vehicle to the point, for example.
I would suggest that you first get a feel for the sensor(s) that you want to use i.e. maybe try computing distances and directions to nearest objects. In real world problems, a major benefit of LiDAR is that it works in any light condition (eg night time) and in most weather conditions, whereas a camera would be limited here due to lack of visibility.

2. The next step might be to do something similar with the road/path you have set up, inside which the vehicle must remain. You might want to think about ways you can check for bounds, so that when the vehicle approaches them, you will know when to stop or slow down appropriately.  More modern methods might use computer vision techniques such as edge detection algorithms to detect eg road lane markings (using camera sensor images).  
For this, you might find some of our Python example scripts useful. 'access_road_network.ipynb' shows how road data can be accessed from BeamNG.tech and plotted. The scripts 'road_definition.py' and even 'object_placement.ipynb' may also be of use.

3. The third step might be to think about rules which you would want your model to have. For example, as a starting point: if your vehicle approaches a wall, what should it do? Maybe you will want it to turn 90 degrees left then try heading in that direction.  This could get as complicated and efficient as you like. A simple general approach might be to turn the car clockwise by some prescribed angle every time it encounters a surface to its front. A more complicated approach might involve predicting further into the future and measuring in multiple directions, so as to find an optimal direction in which to turn at a given time. I recommend you find literature on the subject online and spend some time reading about approaches that other researchers have tried before, then maybe to take some notes on which approaches you find interesting to study. You can think about the properties of a model which interest you, and look at their test results to see how they perform.

Please also be aware that BeamNGPy is a work in progress, and we are constantly adding new features to this interface. We are planning on releasing an update very soon, but it should not affect any of your existing Python code. If it does, it will be minimal. You may also wish to have a look at some of the code we have in Lua, which is where Python gets its data from.

We hope this information helps you with your project, and we wish you success with it.  If you have any further queries in future about the software, we are here to help.

Best regards,
Dave Stark,
Senior Research Software Engineer,
BeamNG.Tech

https://github.com/BeamNG/BeamNGpy/issues/169#issuecomment-1151370291

mayureshb7, 2022-06-20:
Hi Dave,

I apologize for the late response; I appreciate your feedback and great explanation and will incorporate some excellent points you have suggested to me in the simulation/project.
Now I know the software and interface better to implement my work. For the obstacle avoidance course, I will look into how I can successfully use all the sensors at my disposal.

Warm Regards,
Mayuresh Bhosale