BeamNG/BeamNGpy #170: Can't attach camera on linux

GregorKikelj, 2022-06-15:
Trying to run code similar to an example
```
scenario = Scenario('smallgrid', 'Just a_random_scenario')
vehicle = Vehicle('ego_vehicle', model='etki', licence='PYTHON')
cam_pos = np.array([0, 0, 0])  # placeholder values that will be recomputed later
cam_dir = np.array([0, 0, 0])  # placeholder values that will be recomputed later
cam_fov = 70
cam_res = (512, 512)
camera = Camera(cam_pos, cam_dir, cam_fov, cam_res, colour=True)
vehicle.attach_sensor('camera', camera)

scenario.add_vehicle(vehicle, pos=(0, 0, 0))
scenario.make(beamng)
```

However get the error 
```
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
/tmp/ipykernel_4779/33918574.py in <module>
      6 cam_res = (512, 512)
      7 camera = Camera(cam_pos, cam_dir, cam_fov, cam_res, colour=True)
----> 8 vehicle.attach_sensor('camera', camera)
      9 
     10 scenario.add_vehicle(vehicle, pos=(0, 0, 0))

~/.local/share/virtualenvs/openpilot-EZB6qHEP/lib/python3.8/site-packages/beamngpy/vehicle.py in attach_sensor(self, name, sensor)
    234                                 f'with the same name: "{name}"')
    235         self.sensors[name] = sensor
--> 236         sensor.attach(self, name)
    237 
    238     def detach_sensor(self, name):

~/.local/share/virtualenvs/openpilot-EZB6qHEP/lib/python3.8/site-packages/beamngpy/sensors.py in attach(self, vehicle, name)
    531                 self.colour_handle = '{}.{}.{}.colour'
    532                 self.colour_handle = self.colour_handle.format(pid, prefix, name)
--> 533                 self.colour_shmem = mmap.mmap(0, size, self.colour_handle)
    534                 self.logger.debug('Bound shmem for colour: {self.colour_handle}')
    535 

TypeError: an integer is required (got type str)
```

aivora-beamng, 2022-06-15:
Hi, are you running BeamNG.tech on Linux? If so, then shared memory is not working (we are working on a fix). Meanwhile, you can disable shared memory by using the `camera = Camera(cam_pos, cam_dir, cam_fov, cam_res, colour=True, shmem=False)` call.

GregorKikelj, 2022-06-15:
Yeah, trying to get Linux to work. Can close, just wanted to make sure it's a known issue

GregorKikelj, 2022-06-16:
Did some more testing, still can't seem to get the camera to work. Tried running https://raw.githubusercontent.com/BeamNG/BeamNGpy/master/examples/west_coast_random.py with shmem=False but keep getting 
```
Traceback (most recent call last):
  File "/home/gregor/openpilot/tools/sim/testing.py", line 113, in main
    sensors = bng.poll_sensors(vehicle)
  File "/home/gregor/.local/share/virtualenvs/openpilot-EZB6qHEP/lib/python3.8/site-packages/beamngpy/beamng.py", line 1049, in poll_sensors
    vehicle.poll_sensors()
  File "/home/gregor/.local/share/virtualenvs/openpilot-EZB6qHEP/lib/python3.8/site-packages/beamngpy/vehicle.py", line 349, in poll_sensors
    response = self.bng.recv()
  File "/home/gregor/.local/share/virtualenvs/openpilot-EZB6qHEP/lib/python3.8/site-packages/beamngpy/beamng.py", line 524, in recv
    return recv_msg(self.skt)
  File "/home/gregor/.local/share/virtualenvs/openpilot-EZB6qHEP/lib/python3.8/site-packages/beamngpy/beamngcommon.py", line 257, in recv_msg
    length = int(str(length, 'ascii'))
ValueError: invalid literal for int() with base 10: ''
```

aivora-beamng, 2025-01-27:
Closing, Camera sensor has been fixed on Linux since then.

closed 2025-01-27 by aivora-beamng