BeamNG/BeamNGpy #172: Cannot poll IMU sensor data

abhitirumala, 2022-06-16:
Hi,

I am trying to poll sensor data from the IMU in the vehicle, but upon running vehicle.poll_sensors(),  there is a lua error.

My Code:
```
import mmap
import random
import sys

from time import sleep

import numpy as np
from beamngpy import BeamNGpy, Scenario, Vehicle, setup_logging
from beamngpy.sensors import GForces, IMU

# Instantiate BeamNGpy instance running the simulator from the given path,
# communicating over localhost:64256
random.seed(101)
setup_logging()
bng = BeamNGpy('localhost', 64256, home='D:/BeamNG')
# Launch BeamNG.tech
bng.open(launch=True)
# Create a scenario in west_coast_usa called 'example'
scenario = Scenario('west_coast_usa', 'example')
# Create an ETK800 with the licence plate 'PYTHON'
vehicle = Vehicle('ego_vehicle', model='etk800', licence='PYTHON')
imu = IMU()
vehicle.attach_sensor("IMU", imu)

# Add it to our scenario at this position and rotation
scenario.add_vehicle(vehicle, pos=(-717, 101, 118), rot=None, rot_quat=(0, 0, 0.3826834, 0.9238795))
# Place files defining our scenario for the simulator to read
scenario.make(bng)

try:
    # Load and start our scenario
    bng.load_scenario(scenario)
    bng.start_scenario()

    print(vehicle.poll_sensors())
finally:
    bng.close()
```

Lua Error:

[image]

[image]

Thank you!

aivora-beamng, 2022-06-17:
Hi, please specify which version of BeamNG.tech and BeamNGpy you are running.

abhitirumala, 2022-06-17:
Hi, I am using BeamNG.tech version 0.24.0.2 and beamngpy version 1.22

abhitirumala, 2022-06-17:
I am having the same issue when using BeamNG.tech version 0.25 and beamngpy version 1.23

aivora-beamng, 2022-06-21:
Hi, the problem is that you need to specify either the position or the node for the IMU to be spawned at. We will update the documentation to emphasize that requirement.

I edited your example so it is working:

```python
import mmap
import random
import sys

from time import sleep

import numpy as np
from beamngpy import BeamNGpy, Scenario, Vehicle, setup_logging
from beamngpy.sensors import GForces, IMU

# Instantiate BeamNGpy instance running the simulator from the given path,
# communicating over localhost:64256
random.seed(101)
setup_logging()
bng = BeamNGpy('localhost', 64256, home='T:/BeamNG/BeamNG.tech.v0.25.0.0')
# Launch BeamNG.tech
bng.open(launch=True)
# Create a scenario in west_coast_usa called 'example'
scenario = Scenario('west_coast_usa', 'example')
# Create an ETK800 with the licence plate 'PYTHON'
vehicle = Vehicle('ego_vehicle', model='etk800', licence='PYTHON')
imu = IMU(pos=(0.73, 0.51, 0.8))
vehicle.attach_sensor("IMU", imu)

# Add it to our scenario at this position and rotation
scenario.add_vehicle(vehicle, pos=(-717, 101, 118), rot=None, rot_quat=(0, 0, 0.3826834, 0.9238795))
# Place files defining our scenario for the simulator to read
scenario.make(bng)

try:
    # Load and start our scenario
    bng.load_scenario(scenario)
    bng.start_scenario()

    vehicle.poll_sensors()
    print(vehicle.sensors['IMU'].data)
    input('Press Enter to exit...')
finally:
    bng.close()
```