BeamNG/BeamNGpy #174: BeamNGTech access error

mayureshb7, 2022-06-20:
Hi Team,

As instructed in the tech demonstration, I'm able to launch BeamNGTech by exe but unable to launch it through Python API.
I'm getting below mentioned error and need your feedback on it for me to resolve the issue.

[image]

Warm Regards,
Mayuresh Bhosale

aivora-beamng, 2022-06-21:
Hi, your BeamNG.tech and BeamNGpy versions are out-of-sync, please check https://github.com/BeamNG/BeamNGpy#compatibility (you should use BeamNGpy v1.23 with BeamNG.tech 0.25).

mayureshb7, 2022-07-01:
Thank you for your response, I was able to resolve this issue

closed 2022-07-09 by mayureshb7