BeamNG/BeamNGpy #176: beamngpy prompt entry point problem

Mayuemei, 2022-06-28:
I installed BeamNGpy, and then entered beamngpy at the command line, prompting
‘Traceback (most recent call last):
  File "C:\Users\xsm\AppData\Local\Programs\Python\Python39\Scripts\beamngpy-script.py", line 33, in <module>
    sys.exit(load_entry_point('beamngpy==1.23', 'console_scripts', 'beamngpy')())
  File "C:\Users\xsm\AppData\Local\Programs\Python\Python39\Scripts\beamngpy-script.py", line 25, in importlib_load_entry_point
    return next(matches).load()
StopIteration’
I don't think this is correct, what is the problem?Because this problem may cause the following error message when I simulate the scene from other works named Asfault with BeamNg.tech
‘No entry point for level found: /levels/asfault. Ignoring level’
Thanks for your help！

aivora-beamng, 2022-06-28:
Hi,
this is a bug, we do not have a `beamngpy` CLI command anymore in BeamNGpy 1.23. All the functionality is exposed as a Python package, if you want to interact with the library using command line, you can do so by calling the Python interactive mode:
```bash
python -i -c "import beamngpy"
```

Mayuemei, 2022-07-02:
Thank you very much，I have solved this problem.

closed 2022-07-09 by dstark481