BeamNG/BeamNGpy #180: What do the values of some sensors‘ attributes  truely means？

naner34, 2022-07-09:
Hi, When I attach ”electrics“ sensor to a vehicle and try  to get some data, I found  the data I got is not the same as I thought. As follows:

First, the vehicle is define at position (0,0,0)

```python
scenario = Scenario('smallgrid', 'example')
vehicle = Vehicle('ego_vehicle', model='etk800', licence='PYTHON')
electrics = Electrics()
vehicle.attach_sensor('electrics', electrics)
timer = Timer()
vehicle.attach_sensor('timer', timer)
scenario.add_vehicle(vehicle, pos=(0, 0, 0), rot=None, rot_quat=(0, 0, 0.3826834, 0.9238795))
```

Then, I can pull the sensors data and output the data.
```python
sensors = beamng.poll_sensors(vehicle)
timeNow = sensors['timer']['time']
posNow = sensors["state"]['pos']
dirNow = sensors["state"]['dir']
wheelspeed = sensors['electrics']['wheelspeed']
throttle = sensors['electrics']['throttle']
steering = sensors['electrics']['steering']
brake = sensors['electrics']['brake']
```

The part of the output is shown as follows. From here, it can be found that the vehicle's position (-0.     0.001  0.186)  here is not same as what I defined before (0, 0, 0). It really troubles me a lot.   And I also have the following doubts about the meaning of other properties.
1. Is "dir" the moving direction of the vehicle? The scene level I use is "smallgrid" and the ground in this scene looks flat. Why time is 0, the z coordinate value of dir is not 0？
2. Is "wheelspeed" the speed  of the rear wheels?
3. Is the correspondence between "steering" and steering wheel angle linear? 
4. Why is the "brake" value not 0 when t=0 (Vehicle does not start)? 
```
'time', 'pos', 'dir', 'wheelspeed', 'throttle', 'steering', 'brake'
0	[-0.     0.001  0.186]	[-0.    -1.    -0.011]	0.0	0	-0.0	0.3
0.041	[-0.     0.001  0.186]	[-0.    -1.    -0.011]	0.0	0	0.0	0.3
0.081	[-0.     0.001  0.186]	[-0.    -1.    -0.011]	0.0	0	0.0	0.3
0.122	[-0.     0.001  0.186]	[-0.    -1.    -0.011]	0.0	0	-0.0	0.3
0.162	[-0.     0.001  0.186]	[-0.    -1.    -0.011]	0.0	0	-0.0	0.3
0.202	[-0.     0.001  0.186]	[-0.    -1.    -0.011]	0.0	0	-0.0	0.3
0.243	[-0.     0.001  0.186]	[-0.    -1.    -0.011]	0.0	0	0.0	0.3
0.283	[-0.     0.001  0.186]	[-0.    -1.    -0.011]	0.0	0	0.0	0.3
```
Looking forward for a reply. Thans very much.

dstark481, 2022-07-09:
Hi naner34,

Thank you for your queries. I will try to answer them here:

The vehicle position is actually the vehicles 'ref node' snapped to the float grid, which may be slightly different from the position which you said.  The 'ref node' is fairly stable, so if you do not deform the vehicle too much, this will be fixed, and you can take the difference into account if you need to (it should remain the same for all time). Note: even the 'ref node' is not at (0, 0, 0).

Our jbeam engineers choose four stable vehicle nodes to define the local reference system of the vehicle. These nodes should be well-positioned and guarded so as to avoid being exposed to deformation as much as possible. From these four nodes, we compute vectors such as 'forward' and 'up' - these are useful since they provide a local frame for the vehicle at a given time (you can get the third vector by taking the cross product of these two, for example) and use these to form transformation matrices etc.  The reason the direction is not parallel to the XY plane, as you have noticed, is because these vectors will be slightly out.  It is very close to parallel here - -0.011 is very small, and again this should remain constant, so one can take this into consideration when doing any computations with it, if required.  'dir' here is just the forward direction of this frame ie the direction in which the vehicle is facing forward.

'wheelspeed' is usually the average speed of all four wheels.

Steering is linear with the movement within the steering rack, but not necessarily linear with wheel angle (this depends on the suspension geometry, and whether you are looking at the left or right wheel (it will be assymmetric depending on the Ackermann angle setup for the vehicle, eg see https://en.wikipedia.org/wiki/Ackermann_steering_geometry for some more information on that). 

Regarding the question about the brake value:  are you using the 'arcade' shifting mode? this mode will automatically apply some braking for you when the vehicle is stopped.  Maybe try switching to the 'realistic' shifting mode and review the options->gameplay->assistant settings if you do not want the game to be adding extra helper inputs.

We hope the information here will help you somewhat with your queries.

Best regards,
BeamNG Tech Team

naner34, 2022-07-11:
Thank you very much. It helps me a lot.

closed 2022-07-11 by naner34