BeamNG/BeamNGpy #181: Connecting Beamngpy to the simulator

mayureshb7, 2022-07-09:
Hi Team,

I want to use the driving simulator and connect the software to it. When I try running the executable file on the command prompt with --luasocket as an extension, the simulator and game starts running.
But When I try modifying the Beamng.py python script by adding an extension, I get an error as mentioned in the images below.
Can you let me know If I'm adding the extension correctly in the python file or if I need to do something else?

I have tried with both 32-bit and 64-bit, getting the same error.

[image]
[image]
[image]

mayureshb7, 2022-07-09:
[image]
[image]

mayureshb7, 2022-07-09:
In addition to it if I try running it into 32 bit, it throws me following error
[image]

dstark481, 2022-07-09:
Hi mayureshb7,

Thank you for your query.

Could you tell us which version of beamNGpy you are using?  (The latest is v1.23.1).  We will take a look and investigate.  

Looking at your Lua console output: In the latest release , there is no utils/researchGE.  We do have lua/ge/extensions/techcore.lua which superceeds lua/ge/extensions/util/researchGE.lua  - so you could try replacing it with that, and see if it solves the problem (assuming you are using the latest release).

Also please note:  support for 32-bit architectures has been removed in recent releases of beamNG.tech.

Best regards,
BeamNG Tech Team

mayureshb7, 2022-07-09:
Hello Dave,

Thank you for your response!

I'm using 1.22 version of beamngpy which is compatible with BeamNG.Tech.0.24 provided to me.

I'm currently using SimTools Game manager which requires 32-bit process and cannot access 64-bit BeamNG Tech (Getting error mentioned in the below image). 
[image]

Can you please let me know if I can access simulator by any other way without using 32-bit BeamNG.Tech as I was able to use the simulator by launching it on command prompt using "BeamNG.drive.exe --luasocket"

Warm Regards,
Mayuresh Bhosale

aivora-beamng, 2022-07-11:
> [image]
> 
> [image]

Hi, this issue happens because the `-rport` option expects a port number as the next argument. With your code, the arguments of the call look like `BeamNG.drive.x64.exe -console -rport -luasocket 64256 -nosteam`, which does not work.

You need to reorder the options and it should work:
```python
call = [
    self.binary,
    '-console',
    '-rport',
    str(self.port),
    '--luasocket',
    '-nosteam',
]
```

What do you need the `--luasocket` option for? AFAIK, it does not have any effect in the simulator.

mayureshb7, 2022-07-11:
Hi,

I'm using SimTools and driving simulator that works on 32-bit process only. To successfully connect BeamNG.drive game with the 6-DOF driving simulator I execute the command as mentioned in the below.

"BeamNG.drive.exe --luasocket"

[image]

Doing so, BeamNG runs in 64-bit process and driving simulator also works with it:
[image]

Now, I want to add the same extension on Beamngpy and use it to run the simulation code that I wrote, on this driving simulator. I believe by adding --luasocket extension can connect my code to the simulator, correct me if I'm wrong here

As you mentioned, by adding --luasocket after the self.port I'm unable to connect driving simulator to BeamNGpy
[image]

And when I try to run my code I get the following mentioned error which is strange because driving simulator works when I access beamNG from command prompt "BeamNG.drive.exe --luasocket" as mentioned in the image above
[image]

My question is, 
1) How does the 32-bit driving simulator communicates with 64-bit BeamNG game and if I can use the same method to run my code through beamngpy
2) Is there a way in which I can run 32-bit BeamNG game with beamngpy and use driving simulator with it

Warm Regards,
Mayuresh Bhosale

mayureshb7, 2022-07-12:
Hi Adam,

Thank you for your response.

As you mentioned, I tried modifying the executable to a 32-bit process, and BeamNG.drive/tech shuts down when run through python, giving the error mentioned below.

BINARIES = [
    'Bin32/BeamNG.drive.x86.exe',
    'Bin32/BeamNG.tech.x86.exe',
]
[image]

However, I will try using Sim Racing Studio to access the driving simulator as you were successful with it.
Can you please confirm if you were able to use beamngpy with Sim Racing Studio for the driving simulator?

Warm Regards,
Mayuresh Bhosale

aivora-beamng, 2022-07-18:
Hi, I was able to use BeamNGpy with Sim Racing Studio, it should be compatible.

mayureshb7, 2022-07-18:
Hi Adam,

Thank you for your response,  Appreciate it!

I reached out to Sim Racing Studio and currently they don't have support for the simulator that I'm working on.
They will have support on it in two weeks and I will try to use BeamNGpy with SRS for my simulation according to SRS response/support. 

Warm Regards,
Mayuresh Bhosale

aivora-beamng, 2022-07-19:
Great, hope that you will manage to run the simulator with SRS :) If not, feel free to open another issue with your problems.

closed 2022-07-19 by aivora-beamng