BeamNG/BeamNGpy #182: How to use the

naner34, 2022-07-16:


dstark481, 2022-07-18:
Hi,

Thanks for your query.

The tracks are actually separate from the levels, so if you want a track on a level you can load a level then load a track.

Regarding your query 2, this is something we have on our to-do list, but have not got round to yet.  There is currently no way to do this, unfortunately.

Best regards,
BeamNG Tech team.

naner34, 2022-07-19:
Hello dstark481,

Thank you for your response! Looking forward to the fufuture update.

closed 2022-07-19 by naner34

aivora-beamng, 2022-07-19:
Hi again,
if you want to experiment with loading TrackBuilder scenarios before the feature is implemented, you can use the following script:
```python
from beamngpy import BeamNGpy, Scenario, Vehicle
from pathlib import Path

def load_trackbuilder_track(beamng, path):
    file = Path(path).name
    
    lua_code = f'''extensions.load("util_trackBuilder_splineTrack")
local tb = extensions.util_trackBuilder_splineTrack
tb.ignoreCapsForMeshes = true
tb.load(readJsonFile([[{path}]]), true, nil, nil, true, false)

for _, objName in ipairs(scenetree.findClassObjects("ProceduralMesh")) do
  if not string.find(objName, ".json") then
    local obj = scenetree.findObject(objName)
    if obj then
      obj:setName('{file}'..'-'..obj.name)
    end
  end
end
tb.unloadAll()
extensions.unload("util_trackBuilder_splineTrack")
'''
    beamng.queue_lua_command(lua_code)

beamng = BeamNGpy('localhost', 64256)
beamng.open()

scenario = Scenario('smallgrid', 'trackbuilder')
vehicle = Vehicle('ego_vehicle', model='pickup', licence='PYTHON')
scenario.add_vehicle(vehicle, pos=(0, 0, 0), rot_quat=(0, 0, 0, 1))

scenario.make(beamng)
beamng.load_scenario(scenario)
beamng.start_scenario()

load_trackbuilder_track(beamng, r'\levels\gridmap_v2\qwer.json')

beamng.teleport_vehicle(vehicle.vid, (217.36, -262.46, 220.38), rot_quat=(0, 0, 0, 1))
```

naner34, 2022-07-20:
Thank you very much. It really help me a lot. Now I can rebuild the track that has been already saved.