BeamNG/BeamNGpy #184: Changing light bar mode

purple-mesh, 2022-08-09:
In BeamNG, 

* light bar lights canbe toggled using `shift + n` -- and this can be modified using Vehicle class's `set_lights()`
* light bar modes can be changed using `alt + shift + n`

Would it be possible to cycle through light bar modes using BeamNGpy?

aivora-beamng, 2022-08-09:
Hi,
you can run a custom vehicle Lua chunk which does the equivalent of the `alt + shift + n` shortcut. That is,
```python
chunk = '''for _,v in pairs(controller.getControllersByType('lightbar')) do v.toggleMode() end'''
vehicle.queue_lua_command(chunk)
```

where `vehicle` is an instance of a `BeamNGpy.Vehicle` object already connected to BeamNGpy. Let me know if you need further support!

purple-mesh, 2022-08-10:
Thanks @aivora-beamng I have tested this and confirmed that it works as expected. Appreciate the prompt response!

closed 2022-08-10 by purple-mesh