BeamNG/BeamNGpy #185: Using BeamNGpy to control scenario start routine

purple-mesh, 2022-08-16:
Hi,

I was wondering if it is possible to use BeamNGpy to control the scenario start routine including the initial message which is shown to the user.

```
self.beamng.load_scenario(scenario)
self.beamng.start_scenario()
self.beamng.switch_vehicle(vehicle=player_vehicle)
```

For example, when I was testing the above with `BeamNG.tech.v0.25.4.0` sometimes it would show the scenario start message, but other times it would show briefly and go away and give the player direct control of the vehicle without interacting with the GUI.

Would it be possible to:

* configure scenario start message, including title and description?
* configure if the GUI appears or not (i.e. completely disable it)?

aivora-beamng, 2022-08-18:
Hi, the intended behaviour is that the `load_scenario` function loads the scenario and shows the start screen. The `start_scenario` function then hides that start screen and starts the scenario.

If that doesn't happen every time, then it's probably a bug. Can you confirm that the start screen does not close when running the `start_scenario` function?

Configuring the scenario start message title and description is certainly possible. You can try the following code:
```python
scenario = Scenario('west_coast_usa', 'This is the title text', description='This is the description.')
vehicle = Vehicle('ego_vehicle', model='etk800', licence='PYTHON')
scenario.add_vehicle(vehicle, pos=(-717, 101, 118), rot_quat=(0, 0, 0.3826834, 0.9238795))
scenario.make(bng)

bng.load_scenario(scenario)
```

And then you can start the scenario and hide the scren by running `bng.start_scenario()`. As far as I know, the start screen cannot be configured to be completely disabled from BeamNGpy. Is that a feature that you'd want to have? We could try to add it to our roadmap.

purple-mesh, 2022-08-19:
Hi,

Thank you for your clarification. The `start_scenario()` method does explain its behavior and I think it makes more sense now.

```
Starts the scenario; equivalent to clicking the "Start" button in the
game after loading a scenario. This method blocks until the countdown
to the scenario's start has finished.
```

The only small thing which I noticed was that if the player clicks `ESC` key ➡️ pause menu ➡️ `ESC` key again to go back, then the scenario start dialog with `Start` button appears again. I was expecting the `start_scenario()` method to clear that once and for all.

purple-mesh, 2022-08-19:
I also have another question regarding the `Scenario` class, what are `checkpoints`? and how do `checkpoints` differ from `waypoints`?

aivora-beamng, 2022-08-23:
> The only small thing which I noticed was that if the player clicks `ESC` key ➡️ pause menu ➡️ `ESC` key again to go back, then the scenario start dialog with `Start` button appears again. I was expecting the `start_scenario()` method to clear that once and for all.

Ah, I see, thanks for reporting! Unfortunately, this is a bug. It will be fixed in the next major release.

> I also have another question regarding the `Scenario` class, what are `checkpoints`? and how do `checkpoints` differ from `waypoints`?

Hi, the `waypoints` field in the `Scenario` class is unused. To get the waypoints of a scenario, you can call the function `Scenario.find_waypoints()`.

Waypoints are nodes of the navgraph - you can think of them as named positions in a map. If you call `find_waypoints`, you'll see that there are tens of waypoints already predefined on most of the maps.

Checkpoints of a scenario are positions that are needed to be driven through in order to finish a scenario. To create checkpoints in your scenario, you can call the function `Scenario.add_checkpoints` (see the example below).

If you run the example, you can see that every checkpoint created through the `add_checkpoints` function has its associated waypoint.
```python
scenario = Scenario('west_coast_usa', 'waypoint demo')
vehicle = Vehicle('ego', model='etk800')
scenario.add_vehicle(vehicle, pos=(-712.76, 105.05, 118.66), rot_quat=(0.0010, 0.0057, 0.9187, -0.3949))

positions = [(-705.03, 112.40, 118.67), (-696.67, 120.39, 118.65), (-687.03, 129.76, 118.15)]
scales = [(1, 1, 1)] * 3
ids = ['my_waypoint_1', 'my_waypoint_2', 'my_waypoint_3']
scenario.add_checkpoints(positions, scales, ids)
scenario.make(bng)

bng.load_scenario(scenario)
bng.start_scenario()

waypoints = {waypoint.name: waypoint for waypoint in scenario.find_waypoints()}
print(f'{len(waypoints)} waypoints found.')
print(f'Waypoints: [', ', '.join(waypoints.keys()), ']')

vehicle.ai_drive_in_lane(True)
vehicle.ai_set_waypoint('my_waypoint_3')
```

purple-mesh, 2022-08-25:
@aivora-beamng thanks for the explanation and demo.

closed 2022-08-25 by purple-mesh