BeamNG/BeamNGpy #186: IMU Data

mayureshb7, 2022-08-18:
Hi Team,

I would like to know the units of data obtained from IMU sensor, Particularly about ax, ay, az, gx, gy and gz

Warm Regards,
Mayuresh Bhosale

dstark481, 2022-08-18:
Hi Mayureshb7,

Thanks for your query.  

The IMU sensor we currently have is quite basic and will very soon be replaced with a more accurate and physically-based accelerometer sensor.  However, to answer your question: the ax, ay, az are the angular acceleration in each axis - this will be in rads^-2 (radians per second per second), and the gx, gy, gz are the forces measured in each axis (in Newtons).

We hope you will find this information useful.

Best regards,
BeamNG Tech Team

mayureshb7, 2022-08-22:
Thank you for your response Dave; I Appreciate it.

Right now, I'm trying to utilize the vehicle acceleration data as an input for the driving simulator through Python API. IMU data seems to be the one I should be using; correct me if I'm wrong.

I'm curious about two things and would appreciate it if I could get the answer to them.

1) Which sensor data does the driving simulator gets through BeamNG.drive, and can I use the same data via beamngpy/beamng.tech

2) Gforce data that I'm trying to obtain does not run the simulation and I'm getting the following error. Could you please help me out with the same?
[image]

Code:
[image]

Warm Regards,
Mayuresh

dstark481, 2022-08-22:
Hi Mayuresh,

The IMU sensor would the best currently-available sensor to get the acceleration data from, although I would stress that we will shortly be releasing a more advanced accelerometer sensor. The data from that will be more accurate, and you will be able to specify your own coordinate system for it to measure in.  These things are a work in progress, and will improve with better modelling as time goes on.

Regarding the GForces error: what you have shown there looks like a socket bug.  This happens when we try to send data between the simulator and BeamNGpy, and there is a miscommunication between the two. It would be useful to see if there is any corresponding output on the python console window when this error occurs.  If this has happened repeatedly, this is likely a bug at our end which we can investigate.  Regarding the GForces sensor, I am actually not sure what state it is in myself, so I would need to take a look.  It may be incomplete.

Regarding your question about pitch, roll, yaw, heave etc...  The simulator typically deals with more raw geometric quantities such as positions, directions, velocities, accelerations etc.  Certainly pitch, roll and yaw can be computed from looking at changes in the vehicle pose (ie the vehicle position, the forward vector, up vector etc).  The others you describe (translational motions) could be computed with some post-processing, by looking at changes in the vehicle position over time, then resolving them to the vehicles axes (forward, up, right).  This is just a guess - I am no expert on how people typically compute these quantities, or in what units you would want them etc.  For example though, on a flat surface, you might be able to compute the 'heave' of a vehicle simply by looking at how the elevation of the vehicle changes over time.  
The 'sensor' which contains all currently available data like this (position, direction, velocity etc) is called the State sensor, and every vehicle has one of these by default, so when you poll the sensors, the latest data will be available to you in your python script.

Best regards,
BeamNG Tech Team.

mayureshb7, 2022-08-22:
Hi Dave,

Thank you for your response and all the answers.

I will wait for the acceleration sensor and the new update you're bringing.
The GForces error is a frequent error that I get whenever I try to run the code. If I try to do the same with IMU, the code runs well, and I can access the IMU data.

The following error is thrown when I run the code for GForces in a python script and it seems to be a socket error
[image]

Warm Regards,
Mayuresh Bhosale

dstark481, 2022-08-22:
Thanks.  We will take a look at this soon.

Best regards,
BeamNG Tech Team

dstark481, 2022-09-23:
FYI
The new accelerometer sensor is available in our recent release.  Documentation on usage etc can be found here:
https://documentation.beamng.com/beamng_tech/sensors/accelerometer/

Best regards,
BeamNG Tech Team

mayureshb7, 2022-09-26:
Hi Dave,

Appreciate your response and the new accelerometer sensor. I will reach out if there is any other issue.

Warm Regards,
Mayuresh

closed 2022-10-05 by aivora-beamng