BeamNG/BeamNGpy #187: BeamNGpy to centre steering wheel position when scenario begins

purple-mesh, 2022-09-06:
Hi,

I was wondering if it is possible to use BeamNGpy to centre the steering wheel position?

My current setup uses the Logitech G29 steering wheel with force feedback enabled.
Between multiple scenarios, the wheel does not centre itself and the player starts the subsequent scenario with the previous scenario's "finishing steering angle".

So I was wondering if there is any way I can centre the wheel during the scenario setup/starting phase?

aivora-beamng, 2022-09-06:
Hi,

this seems more like a bug. How are you switching between the multiple scenarios? Are you using the in-game functionality or are you using BeamNGpy to switch between scenarios?

purple-mesh, 2022-09-06:
Hi @aivora-beamng,

I am using BeamNGpy to switch between scenarios.

For each scenario in `scenario_0` to `scenario_n` I am using the `beamng.load_scenario()` and `beamng.stop_scenario()` start and stop the scenario.

I think I did not describe the problem fully in my original post. I will try to elaborate a bit further below.

In the first scenario:
* The physical wheel starts centred with `angle = 0`
* Then lets say the player finishes the scenario with wheel `angle = 100 deg`

In the second scenario:
* The physical wheel is on `angle = 100 deg`
* But the in-game player vehicle spawns with the virtual wheel `angle = 0 deg`
* The moment the player interacts with the physical wheel, the virtual wheel angle instantly "changes" to match with the physical wheel angle.

So I was wondering if there is a way I can instruct the Logitech wheel to centre itself after each scenario finishes so that the physical wheel angle will match up with the virtual wheel angle?

aivora-beamng, 2022-09-19:
Thank you for the thorough description, we will look into this issue in the next days.

aivora-beamng, 2022-09-30:
Hi again, unfortunately this is not possible (not only from BeamNGpy, but also not from the simulation), you need to recenter the wheel yourself.

A theoretical possible way of recentering the wheel is using FFB (force feedback) but it would work only if the wheel has FFB, the FFB is strong enough to recenter the wheel and the player does not hold the wheel too strongly.

You can apply FFB from Python using the following function. However, there is no guarantee the force feedback will actually recenter the wheel and you will probably need to tweak the parameters of this function.
```python
from beamngpy import Vehicle

def send_force(vehicle: Vehicle, steps=50000, newtons=20):
    vehicle.queue_lua_command(f'for i=1,{steps} do obj:sendForceFeedback(0, {newtons}) end')
```

purple-mesh, 2022-10-02:
Hi @aivora-beamng, thank you for following up on this.
The Logitech G29 does have force feedback, so I will try it out on the wheel and report back on how it works.

purple-mesh, 2022-11-28:
I had a go by applying the FFB from python using the provided function above on my Logitech G29 setup, unfortunately this is still quite janky and can be a bit too abrupt for users when transitioning between scenarios.

Thanks again for your detailed instruction @aivora-beamng, I will go ahead and close this issue now.

closed 2022-11-28 by purple-mesh