BeamNG/BeamNGpy #188: Poll sensors for multiple vehicles at the same time instance

purple-mesh, 2022-09-19:
Hi,

I was wondering if there is a recommended approach (if any) to perform `poll_sensors()` across multiple vehicles at a particular time instance?

My current thought is to use `bng.pause()` and `bng.resume()` but it doesn't seem like a good option assuming there is a real human player involved.

aivora-beamng, 2022-09-19:
Hi, unfortunately this is not possible with pure BeamNGpy yet. `bng.pause()` and `bng.resume()` is a workaround, but as you say, it limits the player.

The best current solution would be to use a vehicle Lua controller which dumps the data. You can have a look into the `lua/vehicle/controller/` folder of the simulator to see examples of these controllers, namely `lua/vehicle/controller/loggerTemplate.lua`.

If you provide information about what exact data you need I can assist you with creating such a controller and give you instructions how to use it.

purple-mesh, 2022-09-20:
Hi @aivora-beamng,

Thanks for your response.
The data that I am currently gathering are as follows, and I save them to a `.csv` file at the end of each scenario.

```
time
pos (x, y, z)
direction (dir x, dir y, dir z)
wheelspeed 
throttle
steering
brake
damage
damage_ext
vehicle bounding box
```

If I understand your suggestion correctly, do I need add this for each individual vehicle? or does this vehicle Lua controller apply to **all the vehicles** that are added?

aivora-beamng, 2022-09-29:
Hi again,
Unfortunately there is not a simply way to do this that meets your requirements. The problem is that every individual vehicle is an independent Lua process - there is no easy way for timestamps to align at the moment, as the vehicle processes would need a way to synchronize between each other, which would hurt performance. This is a limitation of the engine, and cannot be circumvented easily (one of the possible options is to use the `bng.pause()` and `bng.resume()` methods, but this is not possible for real-time logging).

The "best" way that I can think of is provided below. The timestamps from the main process (which calculates the bounding boxes) and the vehicles' processes still do not match, but the difference between them could be potentially neglected, depending on your concrete usecase.

You can try and see yourself. This example works only for BeamNG.drive/tech 0.26 and later.

1. Download the .zip file [vehicle_logger.zip](https://github.com/BeamNG/BeamNGpy/files/9675232/vehicle_logger.zip) with the mod and copy it to your mods folder (default is `%localappdata%/BeamNG.{drive/tech}/0.26/mods`. You can unpack the mod using the in-game menu to see the included files `vehicleLogger.lua` and `boundingBoxLogger.lua`.
2. Run the following BeamNGpy code (you'll need BeamNGpy 1.24, as it's compatible with BeamNG.tech 0.26):
```python
import time

from beamngpy import BeamNGpy, Scenario, Vehicle

beamng = BeamNGpy('localhost', 64256)
beamng.open()

scenario = Scenario('italy', 'logging_test')

ego = Vehicle('ego', model='etk800', color='White', licence='PYTHON')
ego2 = Vehicle('ego2', model='etk800', color='Red', licence='PYTHON')
scenario.add_vehicle(ego, pos=(245.11, -906.94, 247.46),
                     rot_quat=(0.0010, 0.1242, 0.9884, -0.0872))
scenario.add_vehicle(ego2, pos=(250.11, -906.94, 247.46),
                     rot_quat=(0.0010, 0.1242, 0.9884, -0.0872))

scenario.make(beamng)

beamng.load_scenario(scenario)
beamng.start_scenario()
beamng.switch_vehicle(ego)

ego.ai_set_mode('span')

def start_logging(extension_name='boundingBoxLogger', controller_name='vehicleLogger'):
    command = f'''
        extensions.load('{extension_name}')
        local time = scenario_scenarios.getScenario().timer
        be:queueAllObjectLua(
            string.format([[
                local data = {{time = %d}}
                controller.loadControllerExternal('{controller_name}', '{controller_name}_' .. objectId, data)
            ]], time)
        )
    '''
    beamng.queue_lua_command(command)

def stop_logging(extension_name='boundingBoxLogger', controller_name='vehicleLogger'):
    command = f'''
    be:queueAllObjectLua([[
        controller.getController('{controller_name}_' .. objectId).saveLogToCSV()
        controller.unloadControllerExternal('{controller_name}_' .. objectId)
    ]])
    extensions.unload('{extension_name}')
    '''
    beamng.queue_lua_command(command)

print('Starting logging...')
start_logging()

time.sleep(10)
stop_logging()
print(f'Stopped logging. The logs were saved to {beamng.user}\\0.26.')
```

3. Look at the content of the three `.csv` files that were created.

If you have any questions, then let me know! As you see, the code is not exactly trivial, so in future versions we may provide some helper wrappers to make this whole process easier.