BeamNG/BeamNGpy #189: Can the vehicle in Beamng(v0.25) achieve the same glare intensity as a real-life high beam?

sainan-zhang, 2022-09-21:
Hello, can the vehicle in beamng(v0.25) have high beams that dazzle the eyes? I have set 'headlights=2' in the function 'beamng.set_lights', But the light intensity of the vehicle is not as strong as the high beam in real life. Can you help me achieve this? Looking forward to your reply! Thanks very much!

sainan-zhang, 2022-10-06:
Hi! Thanks for your detailed reply. I have set the value of "lightBrightness" to 1 and created a new vehicle mod according to your suggestion. But the light intensity of the vehicle is still not as strong as the high beam in real life. In the following figure, I copied the blue vehicle, then changed the value of "lightBrightness" to 1.0 to get the red vehicle. That is, the difference between red vehicle and blue vehicle is only the "lightBrightness" value. Is there any other way to make the vehicles in Beamng achieve the same glare intensity as a real-life high beam? Can you give me some help? Looking forward to your reply. Thank you very much!
[image]

aivora-beamng, 2022-10-10:
Hello, you can also increase `lightBrightness` to values higher than 1.0. In this example, I set `lightBrightness` to 10.0.
[image]

I'm afraid achieving a glare similar to real-life is not possible as of now. After additional HDR support is implemented in BeamNG and with a quality HDR screen, the experience should be more similar to real-life lightning.

sainan-zhang, 2022-10-10:
OK, I understand. Many thanks for your kind and warm help!

closed 2022-10-10 by sainan-zhang