BeamNG/BeamNGpy #191: How to get the current time (24-hour format) displayed on the vehicle dial interface?

sainan-zhang, 2022-10-08:
Hi, I found that the vehicle dial interface in Beamng can show the current time in 24-hour format. But I'm confused about how to get this data during the simulation. Can you give me some suggestions? Looking forward to your reply. Thanks very much!
[image]

aivora-beamng, 2022-10-10:
Hi, these vehicle interfaces show the current local time of your computer, so you can get the same time from Python by using the library functions:

```
from datetime import datetime
now = datetime.now()
print(now.strftime("%H:%M"))
```

Be aware that this time is not the time of the environment, which you can see and set in the `Environment` tab in the top menu of the simulation.

sainan-zhang, 2022-10-10:
Hi, it helps me a lot! Thank you very much!

closed 2022-10-10 by sainan-zhang