BeamNG/BeamNGpy #192: How to make the ai vehicle in Beamng start moving at a specific moment

sainan-zhang, 2022-10-13:
Hi, I have a vehicle ‘veh_control’ and an AI vehicle 'veh_ai' in my scenaro.  I want the 'veh_ai 'vehicle starts moving when the veh_control vehicle reaches to a specific position. I tried to use the following code to achieve it, but I failed. Can you give me some suggestions or are there any other ways to realize it? Looking forward to your reply. Thanks!

`scenario = Scenario('smallgrid', 'test')`

`veh_control = Vehicle('ego_vehicle', model='etk800', licence='control')`
`veh_ai = Vehicle('ai_vehicle', model='etki', licence='auto')`

`electrics = Electrics()`
`veh_control.attach_sensor('electrics', electrics)`

`timer = Timer()`
`veh_control.attach_sensor('timer', timer)`

`posControl = (0, -100, 0.4)`
`posAi = (0, -100 + 10, 0.4)`

`scenario.add_vehicle(veh_control, pos=(0, 0, 1), rot=(0, 0, 0))`
`scenario.add_vehicle(veh_ai, pos=(0, 10, 1), rot=(0, 0, 0))`

`scenario.make(beamng)`
`beamng.load_scenario(scenario)`

`beamng.teleport_vehicle(veh_control.vid, posControl, rot=(0, 0, 0))`
`beamng.teleport_vehicle(veh_ai.vid, posAi, rot=(0, 0, 0))`
`beamng.switch_vehicle(veh_control)`
`beamng.start_scenario()`

`def getAiScript(orig=(0, -10, 0),tt=0):
    script = list()
    for i in range(60):
        node = {
            'x': orig[0],
            'y': -i * 30 + orig[1],
            'z': orig[2],
            't': i+tt,
        }
        script.append(node)
    return script`

`timeList = []`
`while True:`
`        sensors = beamng.poll_sensors(veh_cont)`
`        veh_cont.update_vehicle()`
`        pos0 = veh_cont.state['pos']`
`        timeNow = sensors['timer']['time']`
`        if pos0[1] < -100:`
`            timeList.append(timeNow)`
`        tt = timeList[0]`
`        script = getAiScript(posAuto, tt)`
`        veh_auto.ai_set_script(script)`
`        if timeNow > 20:`
`           break`

aivora-beamng, 2022-10-13:
Hello, there were some small issues with the code, try this, it should work as intended by you:
```python
scenario = Scenario('smallgrid', 'test')

veh_control = Vehicle('ego_vehicle', model='etk800', licence='control')
veh_ai = Vehicle('ai_vehicle', model='etki', licence='auto')

electrics = Electrics()
veh_control.attach_sensor('electrics', electrics)

timer = Timer()
veh_control.attach_sensor('timer', timer)

posControl = (0, -100, 0.4)
posAi = (0, -100 + 10, 0.4)

scenario.add_vehicle(veh_control, pos=(0, 0, 1), rot=(0, 0, 0))
scenario.add_vehicle(veh_ai, pos=(0, 10, 1), rot=(0, 0, 0))

scenario.make(beamng)
```
```python
beamng.load_scenario(scenario)
beamng.teleport_vehicle(veh_control.vid, posControl, rot=(0, 0, 0))
beamng.teleport_vehicle(veh_ai.vid, posAi, rot=(0, 0, 0))
beamng.switch_vehicle(veh_control)
beamng.start_scenario()

def getAiScript(orig=(0, -10, 0), tt=0):
    script = list()
    for i in range(60):
        node = { 'x': orig[0], 'y': -i * 30 + orig[1], 'z': orig[2], 't': i+tt, }
        script.append(node)
    return script

timeList = []

while True:    
    veh_control.poll_sensors()
    sensors = veh_control.sensors
    pos0 = veh_control.state['pos']
    timeNow = sensors['timer'].data['time']
    if pos0[1] < -100:
        timeList.append(timeNow)
        tt = timeList[0]
        script = getAiScript(posAi, tt)
        veh_ai.ai_set_script(script)
    if timeNow > 20:
        break
```

If this doesn't work, please let me know the specific issue you are having.

sainan-zhang, 2022-10-13:
Hi, the code works well in the scenario where there is only one ai vehicle and one controlled vehicle. But if I set up two ai vehicles to start moving together when the controlled vehicle reaches a specific position, the problem arises. Two ai vehicles start moving normally, but as time goes on, they end up slowing down to a very slow and audible speed. Then, if I terminate the python process, the two ai vehicles will suddenly speed up, as if they were back to normal speed. I'm exceedingly confused and don't know what's wrong. Besides, I don't know the meaning of each key in script and how it relates to the vehicle movement. Can you help me? My code is shown as follows. Looking forward to your reply. Thanks!

```
from beamngpy import BeamNGpy, Scenario, Vehicle
from beamngpy.sensors import Electrics, Timer

def getAiScript(orig=(0, -10, 0),tt=0):
    script = list()
    for i in range(3600):
        node = {
            'x': orig[0],
            'y': -i * 30 + orig[1],
            'z': orig[2],
            't': i+tt,
        }
        script.append(node)
    return script

beamng = BeamNGpy('localhost', 64256,
                          home=r'D:\software\BeamNG.tech-0.25.0.0.14000\BeamNG.tech-0.25.0.0.14000',
                          user=r'D:\software\BeamNG.tech-0.25.0.0.14000\user')
beamng.open()

scenario = Scenario('smallgrid','test')

veh_cont = Vehicle('ego_vehicle', model='etk800', licence='control')
veh_auto = Vehicle('ai_vehicle', model='etki', licence='auto', colour="Red")
veh_auto2 = Vehicle('ai_vehicle2', model='etki', licence='auto2', colour="Green")

electrics = Electrics()
electrics1 = Electrics()
electrics2 = Electrics()
veh_cont.attach_sensor('electrics', electrics)
veh_auto.attach_sensor('electrics', electrics1)
veh_auto2.attach_sensor('electrics', electrics2)

timer = Timer()
timer1 = Timer()
timer2 = Timer()
veh_cont.attach_sensor('timer', timer)
veh_auto.attach_sensor('timer1', timer1)
veh_auto2.attach_sensor('timer2', timer2)

posCon = (-80, -4000, 0.40)
posAuto = (0, -4000 - 40, 0.40)
posAuto2 = (0, -4000 - 30, 0.40)
scenario.add_vehicle(veh_cont, pos=(0, 0, 1), rot=(0, 0, 0))
scenario.add_vehicle(veh_auto, pos=(0, 10, 1), rot=(0, 0, 0))
scenario.add_vehicle(veh_auto2, pos=(0, 100, 1), rot=(0, 0, 0))

scenario.make(beamng)
beamng.load_scenario(scenario)

beamng.teleport_vehicle(veh_cont.vid, posCon, rot=(0, 0, 0))
beamng.teleport_vehicle(veh_auto.vid, posAuto, rot=(0, 0, 0))
beamng.teleport_vehicle(veh_auto2.vid, posAuto2, rot=(0, 0, 0))
beamng.switch_vehicle(veh_cont)
beamng.start_scenario()

tForBegin = []
while True:
    timeLocalList = []
    veh_cont.poll_sensors()
    sensors = veh_cont.sensors

    veh_auto.poll_sensors()
    sensors1 = veh_auto.sensors

    veh_auto2.poll_sensors()
    sensors2 = veh_auto2.sensors

    pos0 = veh_cont.state['pos']

    timeNow = sensors['timer'].data['time']

    if pos0[1] < -4040:
        tForBegin.append(timeNow)
        script1 = getAiScript0(posAuto,tForBegin[0])
        veh_auto.ai_set_script(script1)
        script2 = getAiScript0(posAuto2,tForBegin[0])
        veh_auto2.ai_set_script(script2)
```

aivora-beamng, 2022-10-14:
Hi, the `poll_sensors()` call blocks the simulation and you are running it continuously in a loop, so it may look like the simulator is lagging. Try to add some delay between the calls, for example using `time.sleep`, so they don't happen so often. Also, after the vehicle passes, you can stop polling the sensors.

```python
import time
tForBegin = []
while True:
    timeLocalList = []
    veh_cont.poll_sensors()
    sensors = veh_cont.sensors

    veh_auto.poll_sensors()
    sensors1 = veh_auto.sensors

    veh_auto2.poll_sensors()
    sensors2 = veh_auto2.sensors

    pos0 = veh_cont.state['pos']

    timeNow = sensors['timer'].data['time']

    if pos0[1] < -4040:
        tForBegin.append(timeNow)
        script1 = getAiScript(posAuto,tForBegin[0])
        veh_auto.ai_set_script(script1)
        script2 = getAiScript(posAuto2,tForBegin[0])
        veh_auto2.ai_set_script(script2)
        break # stop polling sensors when the vehicle passed the location
    time.sleep(0.1) # add delay so `poll_sensors` is not called continuously

input('Press Enter to exit...')
```

You can find information about the vehicle script here: https://beamngpy.readthedocs.io/en/latest/beamngpy.html#beamngpy.vehicle.Vehicle.ai_set_script

It is a list of `(x, y, z, t)` nodes, where `x, y, z` specify the desired location of the vehicle at time `t` (in seconds).

closed 2022-10-14 by sainan-zhang