BeamNG/BeamNGpy #195: What does   "steering"  mean？

naner34, 2022-11-15:
Hello, when I got the "steering" data using fllowed code,
```
steering = sensors['electrics']['steering']
```
I got some data like this.

[image]
[image]

I got some question about the data:
1. What is the unit of data？ Degree？
2. What is the range of data？Maybe from -360°  to 360°
3. Is this the steering wheel angle? If not, how to convert between them.
Maybe the results are obvious, but it's important for me to confirm these. Looking forward for your reply. Thanks a lot.

aivora-beamng, 2022-11-15:
Hello :)

The units are indeed degrees, the range is -720° to 720° and it is the steering wheel angle.

naner34, 2022-11-15:
Thanks，it really helps me a lot.
But the steering wheel angle range of the device I use to play the game  is [- 900°，900°]. 
I wonder if  steering wheel angle of input device is kept at 900 ° or other degree bigger than 720, 
will the  steering wheel angle accepted by the car of the game is - 720 °，
and will the output data be - 720 °? 
Waiting for your reply！

aivora-beamng, 2022-11-16:
Hi again,
the range of [-720°, 720°] applies only to a pickup in a default configuration, sorry for miscommunication.

The `sensors['electrics']['steering']` range is defined on a vehicle basis, every vehicle has a different range. You can find the specific value as the field `steeringWheelLock` in the corresponding `.jbeam` file. For `pickup` in a default configuration, the file you can read this value is `content/vehicles/common.zip\vehicles\common\pickup\pickup_suspension_F.jbeam`.

Be aware that this value is only used for the visual representation of the wheel, it does not have an effect on actual steering.

The steering input from your hardware steering wheel is in `sensors['electrics']['steering_input']` and it is in the range of [-1.0, 1.0], where -1.0 means maximum steer to the left and 1.0 maximum steer to the right.

aivora-beamng, 2023-01-10:
Closing because of inactivity. If you have more questions, feel free to reopen this issue or open a new one.

closed 2023-01-10 by aivora-beamng