BeamNG/BeamNGpy #198: camera/lidar fails to instantiate on linux

naughtyStark, 2022-12-07:
Hi,

the game crashes when I instantiate(not even poll) either a camera or a lidar and set the use_shared_mem=False (meaning that I'm using sockets). I'm using sockets because when running with shared memory python throws the mmap error talked about in this [issue](https://github.com/BeamNG/BeamNGpy/issues/79).
The game runs fine if I don't instantiate either camera or lidar and just poll the state sensors and electrics (script not shown).

beamng.tech version: 0.26
beamngpy version: 1.24
Operating system: ubuntu 20.04
Graphics card: 4x GTX1080Tis
CPU: intel i7 6850 3.6 Ghz 12 core
ram: 62.7 GB

The code runs fine on windows with or without shared memory, so it is likely a Linux compatibility issue. Is there a fix for this yet?
code:
```
import time
from beamngpy import BeamNGpy, Scenario, Vehicle
from beamngpy.sensors import Lidar

def main():
    bng = BeamNGpy('localhost', 64256, home='/home/username/BeamNG/BeamNG', user='/home/username/BeamNG/BeamNG/userfolder')
    bng.open()
    scenario = Scenario('small_island', name="test integration")
    vehicle = Vehicle('ego_vehicle', model='sunburst', partConfig='vehicles/sunburst/race_M.pc')
    scenario.add_vehicle(vehicle, pos=(-67, 336, 34.5), rot_quat=(0, 0, 0.3826834, 0.9238795))
    scenario.make(bng)
    bng.load_scenario(scenario)
    bng.start_scenario()
    time.sleep(10)  # give the game time to load/do its thing
    # NOTE: Create sensor after scenario has started.
    lidar = Lidar('lidar', bng, vehicle, requested_update_time=0.1, is_using_shared_memory=False)     # Send data via shared memory.

    print("starting")
    while True:
        time.sleep()

    lidar1.remove()
    bng.close()
if __name__ == '__main__':
    main()
```

aivora-beamng, 2022-12-08:
Hi, unfortunately Camera and Lidar sensors are not working on Linux hosts yet. The support is planned for the future. We'll include a better error message to inform users about this missing feature.

naughtyStark, 2022-12-18:
Hey, I saw a video for the 0.27 version. Would the camera/lidar bug be fixed in the 0.27 linux version? I love the new desert map btw! Kudos to the devs for putting that in the game!

aivora-beamng, 2023-01-10:
Unfortunately not. We will focus on implementing the support of sensors for Linux hosts for the next updates, but I cannot provide you with a 100% confirmation it will be fixed in the next version.

MoreTore, 2023-02-22:
Any update on linux support? Need it to work in order to use it.

AbdelrahmanElsaidElsawy, 2024-01-08:
Hello, 
I have a temporary solution for the Linux support (if you have windows machine + BeamNG.Tech), you can use WLS2 as your linux environment, follow the following steps to run any example from BeamNGpy in Linux.

1- run the BeamNG.Tech in windows using command prompt, use the following command `Bin64\BeamNG.drive.x64.exe -rport 64256 -nosteam -lua registerCoreModule('tech/techCore')`
2- run your script in your WSL2 environment 
this is the code i used for the test (i used a modified version of @sidtalia's code)
3- update the script with correct IP address of your windows host 
 
```
import time
from beamngpy import BeamNGpy, Scenario, Vehicle
from beamngpy.sensors import Lidar

def main():
    bng = BeamNGpy('192.168.1.105', 64256)
    bng.open(listen_ip='*' ,launch=False, deploy=False)

    scenario = Scenario('small_island', name="test integration")
    vehicle = Vehicle('ego_vehicle', model='etk800', licence='PYTHON')
    scenario.add_vehicle(vehicle, pos=(-67, 336, 34.5), rot_quat=(0, 0, 0.3826834, 0.9238795))
    scenario.make(bng)
    bng.load_scenario(scenario)
    bng.start_scenario()
    time.sleep(10)  # give the game time to load/do its thing
    
    # NOTE: Create sensor after scenario has started.
    lidar = Lidar('lidar', bng, vehicle, requested_update_time=0.1, is_using_shared_memory=False)

    print("starting")
    while True:
        time.sleep(1)

    lidar.remove()
    bng.close()

if __name__ == '__main__':
    main()
```

[image]

sidtalia, 2024-01-08:
Hi,
Thank you for the response. However, this doesn't satisfy the requirement. 
The issue is that many of my collaborators don't have spare Windows machines to run the simulator on, and so can not follow the WSL approach. 
Again, I urge that the devs look into a way of getting camera and lidar on Ubuntu, even if it happens over socket instead of through shared memory, at least as a starting point until we get shared memory. 
Thanks,Regards,Sidharth Talia.

    

Screenshot.2024-01-08.115125.png (view on web)

aivora-beamng, 2025-01-27:
Recent versions of BeamNG.tech include support for sensors on Linux, closing.

closed 2025-01-27 by aivora-beamng