BeamNG/BeamNGpy #199: Short draw distance from Camera with version 0.27

Diminati, 2022-12-18:
Hello,
I am using BeamNGpy to generate a video from a forward looking camera mounted on the ego vehicle, for a personal project.
I noticed that the camera provides such a short draw distance; most of the graphic elements are popping up a lot later than I would like.

[image]

I upgraded to 0.27 from 0.24, I didn't have this issue in the past. I couldn't correlate this to any update in the Changelog.
Is there any setting to adjust this rendering distance?
I am willing to sacrifice a little bit of performance to get a nicer image.

Thank you.

dstark481, 2022-12-18:
Hi Vincent,

The camera contains both a near and far plane value (in metres), which describe the minimum and maximum range which the camera can 'see'.  Here, it looks like the far plane of the camera is too short.  

Currently, as default values, we have `near_far_planes: Float2 = (0.05, 100.0)`.  You can see this for yourself if you look at the constructor in `camera.py`.  

However, you can increase the far plane (the 2nd number) to something like 1000.  For example, when you set the camera up, include `near_far_planes=(0.05, 1000.0)` in the argument list of the function call.  

We will increase this default value for the next release.  This was most likely a typo during coding.

Best regards,
BeamNG Tech Team

Diminati, 2022-12-18:
Hello Dave,
Thank you very much for your detailed and quick response, modifying near_far_planes did fix my issue.

I think the argument description in the camera class could be a bit more clear.

Great support as always!
Regards,
Vincent

closed 2022-12-18 by Diminati