BeamNG/BeamNGpy #203: beamngpy setup-workspace not working in version 1.21.1

hildebrandt-carl, 2023-02-20:
Hi,

I am trying to run older versions of BeamNG (I need to do this for an artifact submission). The version I was using still required users to setup a workspace manually (whoever decided to stop doing that in the new versions, I owe you a coffee).

Specifically, I am using:
BeamNG version: `BeamNG.tech.v0.23.5.1`
BeamNGpy version: `1.21.1`

I was specifically following the [README](https://github.com/BeamNG/BeamNGpy/tree/v1.21.1#prerequisites) and my [old answer](https://github.com/BeamNG/BeamNGpy/issues/146#issuecomment-1003394130_).

Basically, it states that you need to:
1) Create an environment variable called `BNG_HOME` that points to your version of `C:\<path>\BeamNG.tech-0.23.5.1`.
2) Run the command `beamngpy setup-workspace C:\<path>\BeamNGWorkspace\` that points to your `BeamNGWorkspace`.

I have confirmed that I have done part 1 correctly as I can see it when I list all environment variables using `gci env:* | sort-object name`. However when I run part 2 I get the following error:
```
PS C:\Users\hilde> beamngpy setup-workspace C:\Users\hilde\BeamNG\BeamNGWorkspace\
beamngpy : The term 'beamngpy' is not recognized as the name of a cmdlet, function, script file, or operable program.
Check the spelling of the name, or if a path was included, verify that the path is correct and try again.
At line:1 char:1
+ beamngpy setup-workspace C:\Users\hilde\BeamNG\BeamNGWorkspace\
+ ~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (beamngpy:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException
```

Any pointers would be greatly appreciated :)

hildebrandt-carl, 2023-02-20:
@aivora-beamng thanks for the quick answer, now I owe you 2 coffees ☕  

So with your answer and tinkering I managed to get it working, however as you will see my solution is a bit of a hack. I am hoping you can point out where I went wrong. If not that's okay, if its not broke don't fix it right (other than this manual workspace thing, really again I am so happy you guys changed that 😄)

I repeated your steps of creating a virtual env, activated it, installed beamngpy==1.21.1, and ran the setup-workspace command. However, I still was running into the error where it does not recognize `beamngpy`.

However I noticed a warning (which I admittedly previously ignored, silly me) while installing `beamngpy==1.21.1`:
```bash
WARNING: The script beamngpy.exe is installed in `C:\Users\hilde\AppData\Roaming\Python\Python310\Scripts` which is not on PATH.
```

So I took a guess and went to that directory and saw `beamngpy.exe`. I was then able to setup the workspace using that executable.

So it would seem that I have not set my path up correctly, although I don't recall reading anything about it. Right now I have `BNG_HOME` set to my local copy of `beamng.tech-0.23.5.1`. Should I create another system variable pointing to the `beamngpy.exe`, so that I don't have to manually go into `Roaming\Python\Python310\Scripts`?

aivora-beamng, 2023-02-20:
Ah OK, that's weird, you probably have a little bit different installation of Python than I do. If you add `C:\Users\hilde\AppData\Roaming\Python\Python310\Scripts` to your `PATH` Windows environment variable, then you won't have to manually go to that folder.

hildebrandt-carl, 2023-02-20:
Thank you! Seems like that fixed the problem.

closed 2023-02-20 by hildebrandt-carl