BeamNG/BeamNGpy #205: Vehicle collision cannot be disabled

BlackStar7713, 2023-02-22:
Hello,
I'm trying to disable collision between vehicles via beamngpy.

In BeamNG the setting is Options->Gameplay->Simplified collision physics.
When changing this setting, the settings.json file in the user folder gets the new line ""disableDynamicCollision":"true""

Using the settings.change("disableDynamicCollision", "true") method works, as long as the disableDynamicCollision value is not already in the settings.json file.
If the value exists and is set to false, the method changes the setting in the json file, but in game the collisions are still enabled
(On further testing, this is not the deciding factor wether it works or not, as just now it failed to apply the settings even after deleting the line before starting)

In the in game settings menu, the simplified physics option is not selected, it does however show the warning below the setting, that normally only shows up when the setting is enabled.

This is my testing script, it has so far always reproduced the error on the second launch:

```
import time
from beamngpy import BeamNGpy, Scenario, Vehicle

bng = BeamNGpy('localhost', 64256, home='C:/Users/cryst/Documents/beamng/tech', user='C:/Users/cryst/Documents/beamng/user')
bng.open()
scenario = Scenario('smallgrid', 'example')
vehicle = Vehicle('ego_vehicle', model='etk800', license='PYTHON')
scenario.add_vehicle(vehicle, pos=(190, 0, 0), cling=True)
vehicle2 = Vehicle('vehicle2', model='etk800', license='PYTHON2')
scenario.add_vehicle(vehicle2, pos=(180, 0, 0), cling=True)

scenario.make(bng)
bng.settings.change("disableDynamicCollision", "false")
bng.scenario.load(scenario)
bng.scenario.start()
time.sleep(5)
bng.close()

time.sleep(5)

bng.open()
scenario = Scenario('smallgrid', 'example')
vehicle = Vehicle('ego_vehicle', model='etk800', license='PYTHON')
scenario.add_vehicle(vehicle, pos=(190, 0, 0), cling=True)
vehicle2 = Vehicle('vehicle2', model='etk800', license='PYTHON2')
scenario.add_vehicle(vehicle2, pos=(180, 0, 0), cling=True)

scenario.make(bng)
bng.settings.change("disableDynamicCollision", "true")
bng.scenario.load(scenario)
bng.scenario.start()
```

aivora-beamng, 2023-02-22:
Hi, I will check the problem with settings, but in the meantime, you should be able to call:

```python
bng.control.queue_lua_command('be:setDynamicCollisionEnabled(false)')
```

Does it work for you?

BlackStar7713, 2023-02-22:
Hi,
Unfortunately replacing the previous commands with the lua command did not disable collisions either.

AbdelrahmanElsaidElsawy, 2025-02-21:
You can enable ghost mode using:  
```lua
vehicle.queue_lua_command('obj:setGhostEnabled(true)')
```  
Let us know if you need further assistance!  

---