BeamNG/BeamNGpy #206: Synchronizing Sensor Polling with Simulation

hildebrandt-carl, 2023-03-01:
Hi,

I am using the following versions:
BeamNG.tech: `v0.27.2.0`
BeamNGpy: `1.25.1`

I spawn a vehicle with 3 front-facing cameras and an electronics sensor. My goal is for the vehicle to drive around the map using the `span` ai mode. While it is driving around, I want to capture camera data, steering, and throttle data at a set frequency (4Hz). To do this, I use the following code:

```python
import time
import random

import matplotlib.pyplot as plt

from tqdm import tqdm

from beamngpy import BeamNGpy
from beamngpy import Scenario
from beamngpy import Vehicle
from beamngpy import set_up_simple_logging

from beamngpy.sensors import Camera
from beamngpy.sensors import Electrics

def initialize_cameras(bng, vehicle):
    camera_resolution=(200,150)
    # Create camera dnd lidar after starting simulation
    camera_l = Camera(name="camera_left",
                      bng=bng,
                      vehicle=vehicle,
                      requested_update_time=0.2,
                      dir=(0.75, -1, 0),
                      resolution=camera_resolution)
    camera_m = Camera(name="camera_middle",
                      bng=bng,
                      vehicle=vehicle,
                      requested_update_time=0.2,
                      dir=(0, -1, 0),
                      resolution=camera_resolution,
                      near_far_planes=(0.5, 400))
    camera_r = Camera(name="camera_right",
                      bng=bng,
                      vehicle=vehicle,
                      requested_update_time=0.2,
                      dir=(-0.75, -1, 0),
                      resolution=camera_resolution)
    # Return the cameras            
    return [camera_l, camera_m, camera_r]               

def initialize_electrics(vehicle):

    # Create the electrics
    electronic_sensor = Electrics()
    vehicle.sensors.attach('electrics', electronic_sensor)

    # Return the cameras
    return [electronic_sensor]          

def plot_cameras(data_l, data_m, data_r):
    # Show the camera data
    try:
        plt.figure(1, figsize=(10,3))
        ax = plt.subplot(1,3,1)
        ax.axis('off')
        plt.imshow(data_l)  

        ax = plt.subplot(1,3,2)
        ax.axis('off')
        plt.imshow(data_m)  
        
        ax = plt.subplot(1,3,3)
        ax.axis('off')
        plt.imshow(data_r)  

        plt.subplots_adjust(wspace=0, hspace=0, left=0, right=1, bottom=0, top=1)

        plt.pause(0.001)
    except Exception as e:
        print("{}".format(e))
    return plt

def main():

    # Set capture duration in seconds
    capture_time = 100 
    # Set the capture rate in Hz
    capture_rate = 4
    capture_frequency = 1.0 / capture_rate

    # Generate a new random seed
    random.seed(1703)
    set_up_simple_logging()

    # Declare the BeamNG home and workspace
    bng_home = 'C:/Users/hilde/Desktop/SelfDriving/BeamNG.tech.v0.27.2.0'
    bng_user = 'C:/Users/hilde/Desktop/SelfDriving/BeamNGWorkspace'

    # Generate a new instance of BeamNG
    beamng = BeamNGpy('localhost', 64256, home=bng_home, user=bng_user)
    bng = beamng.open(launch=True)

    # Create the scenario
    scenario = Scenario('west_coast_usa', 'data_capture', description='Spanning the map with a LiDAR sensor')

    # Create and add ego ego_vehicle
    ego_vehicle = Vehicle(vid='ego', model='etk800', license='RED', color='Red')

    scenario.add_vehicle(vehicle=ego_vehicle,
                         pos=(-717.121, 101, 118.675),
                         rot_quat=(0, 0, 0.3826834, 0.9238795))
    scenario.make(bng)

    # Set simulator to 60hz temporal resolution
    bng.settings.set_deterministic(60) 

    # Load simulation
    bng.scenario.load(scenario)
    bng.ui.hide_hud()
    bng.scenario.start()

    # Create the cameras
    cameras = initialize_cameras(bng, ego_vehicle)
    camera_l = cameras[0]
    camera_m = cameras[1]
    camera_r = cameras[2]    

    # Create the electrics
    electrics =initialize_electrics(ego_vehicle)
    elec_sensor = electrics[0]

    # Set vehicle tro drive around randomly
    ego_vehicle.ai.set_mode('span')

    # Drive around
    for i in tqdm(range(int(capture_time / capture_frequency))):

        # Get the latest readings
        data_l = camera_l.poll()
        data_m = camera_m.poll()
        data_r = camera_r.poll()

        # Poll the electronic sensors
        ego_vehicle.sensors.poll()
        print("steering: {}".format(elec_sensor['steering']))
        print("steering_input: {}".format(elec_sensor['steering_input']))
        print("")
        print("throttle: {}".format(elec_sensor['throttle']))
        print("brake: {}".format(elec_sensor['brake']))
        print("--------------------")

        # Show the data
        plot_cameras(data_l['colour'], data_m['colour'], data_r['colour'])

        # np.save("./data/data_l{:04}".format(i), data_l['colour'])
        # np.save("./data/data_m{:04}".format(i), data_m['colour'])
        # np.save("./data/data_r{:04}".format(i), data_r['colour'])

        # Maintain the rate
        time.sleep(capture_rate)

    # Remove sensors
    camera_l.remove()
    camera_m.remove()
    camera_r.remove()

    # Close BeamNG
    bng.close()

if __name__ == '__main__':
    main()
```

 When I ran my script, I noticed that the sensors did not record data anywhere near the set frequency, as you can see below (here I only get 2-3 camera readings in 10 seconds of driving):

[image]

I tried the following to improve the rate at which the sensors poll:
1) Save the camera data rather than plotting it
2) Reducing the camera resolution significantly
3) Using only a single front-facing camera

While some worked better than others, the sensors were always slower than the actual simulation. While I know, there are certain things I can do to my script (for example, tracking the time of the main loop and only waiting for the remaining time, etc.). I was wondering if there is a blocking function I can call. Something that would look as follows:

```
While driving:
    poll sensors
    wait for sensor readings
    continue simulation
```

Otherwise, I am open to other suggestions :) Thanks!

dstark481, 2023-03-01:
Hi Carl,

In your Python main loop you are pausing Python for 4 seconds each time.  This is why you are only getting 2.5 images per 10 second period.  Its this:  time.sleep(capture_rate) where capture_rate is a constant set to 4 at the top of the script.  If you remove this it updates faster.

Perhaps there is a misunderstand between what it means to 'poll' and what the 'update rate' is?  The update rate is a parameter we set which tells the physics engine how often to actually send requests to the GPU for new camera data.  Polling is when you want to send collect that new data from the physics engine and have it available for you in python, which is on a 4 second schedule in your script.

You have set the camera update rate to 0.2 seconds, which means the camera (internally) will automatically get itself new data every 0.2 seconds (on average - these jobs are actually scheduled for optimal GPU processing).  But in Python, you don't actually 'get' the data until you poll the camera, and this has a 4 second delay between each time in your script there.

If you want even faster streaming of the data, even at high resolutions, we currently recommend to switch the shared memory on (in the Camera constructor).   We can do that at least at 20 times per second (at least, with 1 camera).  You shouldn't need that for these resolutions, though.

Best regards,
Dave

hildebrandt-carl, 2023-03-01:
Thank you, using`capture_rate` instead of `capture_frequency` was very silly of me.

However, its good to know that you are able to capture camera data up to `20Hz` using 1 camera. As you said, I found using `is_using_shared_memory=True` to be a huge help. I also found that my plotting function was taking up to a second to render the images which is another reason I was seeing lag.

Thank you.

closed 2023-03-01 by hildebrandt-carl