BeamNG/BeamNGpy #207: Changing LiDAR 'vertical_angle' fails to reflect inside BeamNG

MatousKundrik, 2023-03-11:
Supplying a `vertical_angle` paramater to virtual LiDAR fails to reflect inside BeamNG. I am using the `west_coast_lidar.py` example and have set my LiDAR up as so `lidar = Lidar('lidar1', bng, vehicle, requested_update_time=0.01, is_using_shared_memory=True, pos = (0, 0, 5.7), vertical_angle=75.0)` with the only changes from the original example being the LiDAR is placed 4 units higher than default and the FOV has been increased to 75.0. This yields the following result:

[image]

Here you can see the LiDARs `vertical_angle` appears to remain at the default `26.9 degrees` with a large 'dead zone' around the vehicle. If I log the args being passed to `createLidarWithSharedMemory` or `createLidar` inside `./lua/ge/extensions/tech/sensors.lua` I can see that  the following, seemingly correct, values are being supplied within `args`:

`Args:, { ["colourShmemName"] = 49404.lidar1.Colour,["colourShmemSize"] = 400000000,["pointCloudShmemName"] = 49404.lidar1.PointCloud,["pointCloudShmemSize"] = 1200000000,["verticalResolution"] = 64,["isVisualised"] = true,["isStatic"] = false,["raysPerSecond"] = 2200000,["isForceInsideTriangle"] = false,["horizontalAngle"] = 360,["maxDistance"] = 120,["pos"] = vec3(0,0,5.7),["isSnappingDesired"] = false,["isAnnotated"] = false,["dir"] = vec3(0,-1,0),["frequency"] = 20,["up"] = vec3(0,0,1),["requestedUpdateTime"] = 0.01,["verticleAngle"] = 75,["updatePriority"] = 0,}`

However if I then manually force an update inside the files like so:
```lua
local function createLidar(vid, args)
  args["verticalAngle"] = 75.0
  return Research.SensorManager.createLidarSensorWithoutSharedMemory(vid, args)
end

local function createLidarWithSharedMemory(vid, args)
  args["verticalAngle"] = 75.0
  return Research.SensorManager.createLidarSensorWithSharedMemory(vid, args)
end
```
The correct `vertical_angle` is reflected inside BeamNG as can be seen here (horizontal FOV increased resulting in a smaller 'dead zone'):

[image]

I'm currently using `BeamNG.tech.v0.27.2.0`, as I recall this issue wasn't present in version `BeamNG.tech.v0.23.5.1` and potentially not even as late as `BeamNG.tech.v0.25.0.0`

dstark481, 2023-03-11:
Hello Matous,

You have found a small bug.  Its a spelling error in the `ge/techCore.lua` file:
`args.verticleAngle = request['vAngle']`
`verticle` should be `vertical`.

This is what causes the trouble.  A table of arguments is created ('args'), then in the engine we check for the existence of string names for each possible argument.  If 'args' contains the string, we use its corresponding value to set the parameter in the engine (where the sensor truly lives).  If the string does not exist, we use a default.  Because of the spelling mistake, the engine thinks you are not supplying a value for that vertical angle parameter.

We will update this for the forthcoming release.  Until then, you could manually fix the spelling mistake in the above-mentioned file.

Best regards,
BeamNG Tech Team

MatousKundrik, 2023-03-12:
Thanks for the quick reply @dstark481, will patch locally until next release. 👍

closed 2023-03-12 by MatousKundrik