BeamNG/BeamNGpy #208: Is there any HD map (e.g. OpenDrive) for BeamNG?

CAS-LRJ, 2023-03-13:
Hello, I appreciate your great effort to make BeamNG a more realistic simulator. Now I am using it as an simulator for autonommous driving, and I am wondering that if you can provide some HD map files (e.g. opendrive xodr format) to benefit the route planning. Currently, I am using the lua mod #86 to generate global planning, it works but not perfect (the plan is not optimized). For local planning, an alternate method is to use the road definition to draw the local map, however it is not perfect since some roads are not designed as DecalRoad (e.g. tunnels and highway gateways).

CAS-LRJ, 2023-04-26:
Get it from the new update! Thanks so much!

closed 2023-04-26 by CAS-LRJ

dstark481, 2023-04-26:
The OpenDrive exporter is a work in progress, but it should provide any of our maps in .xodr format.  There are some limitations with regards to lanes etc.  Please let us know if you encounter any issues with it.  Any feedback is appreciated.

Best regards,
BeamNG Tech Team