BeamNG/BeamNGpy #211: False LiDAR Data

CAS-LRJ, 2023-03-30:
The LiDAR receives false data in front of a tunnel when the ego vehicle was driving in the Italy map.
Level: Italy
Position: (1535.8941650390625, 793.4702758789062, 149.81668090820312)

In-Game LiDAR Visualization
[image]

Recevied LiDAR Visualization (Blender)
[image]

dstark481, 2023-03-30:
What is plotted in the Blender plot?  I am not sure what you are comparing here.

CAS-LRJ, 2023-03-31:
> What is plotted in the Blender plot? I am not sure what you are comparing here.

I use a python script to export the readings to a .obj file
```
readings_data = lidar.poll()
data1 = readings_data['pointCloud'].reshape(-1, 3)
# Remove Null Data like (0,0,0)
data1_ind = np.linalg.norm(data1, axis=1) > 0.
data1 = data1[data1_ind]

# Move the coordinates
data2 = np.array(lidar.get_position())
pc_obj = data1 - data2

with open('example.obj', 'w') as f:
    for point in pc_obj:
        f.write('v '+str(point[0])+' '+str(point[1])+' '+str(point[2])+'\n')
```

Then I visualize the obj file in Blender, and I found that there exist some abnormal points (in white circle).
These floating points shoud not be here since there is no obstacles in front of the lidar.
[image]

dstark481, 2023-03-31:
Okay.  We will take a look.

Quant14, 2025-12-15:
Hi, we haven't been able to replicate these issues in the latest version of BeamNG.tech. Please feel free to create a new issue or question on the [forum](https://forum.beamng.tech/) if you have trouble again.

closed 2025-12-15 by Quant14