BeamNG/BeamNGpy #214: the object "cloud layer" in world editor can not be saved in the level of "small grid"

sainan-zhang, 2023-05-08:
hello,
I want to create a new level based on the level of "small grid". I try to add the "cloud layer" object into the level of "small grid" through world editor. However, even if I have clicked the save button, there is no "cloud layer" in the scene when I next enter the level. I found that when I created "cloud layer" object, the event tree window did not change and there was no display of adding "cloud layer". I'm very confused. Can you give me some suggestions? Looking forward to your reply! Thanks a lot!

sainan-zhang, 2023-05-08:
I have solved this problem by creating "cloud layer" in other classes in world editor. But is there a way to add the sound of rain?

sainan-zhang, 2023-05-08:
I have downloaded a mod named "flood", which can add rain and the sound of it.  Can I use beamngpy to call this mod and set the precipitation parameters in the mod through code? Instead of manually setting and adding precipitation after the scene is loaded. I don't know how to realize it through beamngpy. Can you help me? Thanks very much!

dstark481, 2023-05-09:
Hi Sainan,
Thanks for your query.
We do not support third party mods unfortunately so we cannot answer this, but this is the repo for BeamNGpy - BeamNG's Python front end.
You may be able to find an answer to your question on the forums here, however:
https://www.beamng.com/forums/

Best regards,
BeamNG Tech Team

closed 2023-07-24 by aivora-beamng