BeamNG/BeamNGpy #223: Unable to obtain any return value when I use mesh.get_node_positions()

yuyuyu223, 2023-06-25:
### The nodes are empty, so I cannot know the location of the nodes, but the triangle can be obtained
[image]

### The following error occurred in the console:
```
025.735|E|libbeamng.extensions|extension unavailable: "tech_mesh" at location: "tech/mesh"
025.736|E|libbeamng.lua.V|*** FATAL LUA ERROR: [string "string"]:1: attempt to index field 'tech_mesh' (a nil value)
025.737|E|libbeamng.lua.V|=============== Stack Traceback >> START >>
025.737|E|libbeamng.lua.V|(1) main chunk of string at line 1
025.741|E|libbeamng.lua.V|--------------- << END <<
025.742|E|libbeamng.lua.V|*** while executing this line: extensions.tech_mesh.create("a{3B13GFXUpdateTime-0A8sensorId-0B17physicsUpdateTime-0.01~")
025.743|E|libbeamng.lua.V|*** in chunk: string
```

aivora-beamng, 2023-06-26:
Hi, what are your versions of BeamNGpy and BeamNG.tech?

yuyuyu223, 2023-06-28:
> Hi, what are your versions of BeamNGpy and BeamNG.tech?

BeamNG.tech.v0.28.2.0
beamngpy 1.26

dstark481, 2023-07-02:
Hi,
There is some trouble with this feature at the moment due to some missing functions in the back end.  We will have it fixed soon.
Best,
BeamNG Tech Team

aivora-beamng, 2023-10-06:
Hi, the Mesh sensor is fixed in BeamNG.tech 0.30 and BeamNGpy 1.27, please update to the latest version to get rid of this problem.

closed 2023-10-06 by aivora-beamng