BeamNG/BeamNGpy #224: Controlling vehicle actuation with road wheel angle and acceleration.

BleedObsidian, 2023-06-29:
Hello,

Currently the only way to control the actuation of a vehicle is using a steering wheel, throttle and brake interface. This type of interface is very much tied to the exact make and model of car you're using:

1. The steering rack ratio of every car is different (and often not linear). The maximum achievable road wheel angle at full lock of the steering wheel also differs between cars.
2. 50% throttle on a more powerful car is likely to result in a faster rate of acceleration. To deal with this, I have to write PID controllers that have to be tuned differently for every car I use in the simulator.

A more common interface (found on drive-by-wire systems on a real car) is to control using:

- Road Wheel Angle in degrees or radians for lateral control.
- Acceleration in m/s^2 for longitudinal control (positive for forward acceleration, negative for braking).

Any chance we could get this interface type of interface? I think it's a really important issue personally.

dstark481, 2023-06-29:
Thanks Jesse,

We can consider it.

Regarding the first point, you could get the two front road wheel angles (RWA) with something like this in Lua:
```
-- Get the two RWA angles from the vehicle (FL and FR).
local function getWheelAngles()
  local steeringSign = sign(electrics.values.steering_input)
  local frontLeftWheel = wheels.wheelRotators[wheels.wheelRotatorIDs["FL"]]
  local frontRightWheel = wheels.wheelRotators[wheels.wheelRotatorIDs["FR"]]
  local frontLeftWheelAngle = acos(obj:nodeVecPlanarCosRightForward(frontLeftWheel.node2, frontLeftWheel.node1)) * steeringSign
  local frontRightWheelAngle = acos(obj:nodeVecPlanarCosRightForward(frontRightWheel.node1, frontRightWheel.node2)) * steeringSign
  return frontLeftWheelAngle, frontRightWheelAngle
end
```
you also know the steering wheel angle (STWA) in the range [-1, 1], since it is whatever you set it to.
From there, you can compute the map between STWA and RWA, which as you say will depend on the steering mechanism of the vehicle, and will not be linear.  This may be a bit tedious, but it would only need done once per vehicle.  You could save the data to .csv or whatever.

On the second point regarding acceleration, though, I am not sure it is completely clear how to do this well.  It sounds like some kind of PID or similar.  We can either supply a gas pedal value in [0, 1] or supply the propulsion torques directly, but it sounds like we would need some kind of feedback thing where we monitor the acceleration value and adjust accordingly, which will be pretty laggy.  We would need to investigate how the drive-by-wire systems do it, then.  The relationship between acceleration and pedal/torques would be complicated, so you cant just do a look-up or have some formula, I guess.

Best regards,
Dave

BleedObsidian, 2023-06-29:
Hi Dave,

The lua code is really useful thank you, it would be really nice if this was integrated natively so we can get the road wheel angles through `vehicle.state` or similar. At least this is a step upwards.

As for the second point, yeah ultimately real drive by wire systems will be using PID (or potentially LQR or MPC) controllers. The main thing is dealing with wheel slip under heavy acceleration or low friction surfaces, so no a lookup table doesn't cut it really. But you mentioned you can control via propulsion torque, again this would be really useful to expose via the Python interface as torque is far less tied to the vehicle than gas pedal position and it's one step further to controlling acceleration directly (if you assume no tyre slip).

Thanks,
Jesse.

dstark481, 2023-06-29:
Controlling the vehicle through torques (drive and brake) alone is something we already do through Simulink, eg see https://github.com/BeamNG/BeamNG-Simulink_generic_interface,
but would take some work for BeamNGpy.  Let me discuss it, it might make a good little project for someone.
Getting the RWAs through in the vehicle state should be simple enough.

Best,
Dave

dstark481, 2023-06-29:
I have had trouble with that function before, but maybe you will have better luck.  I would be sure to test it out carefully first.

aivora-beamng, 2024-05-14:
Hi, the steering wheel angle is just visual, as reported in the comment. Support for RWA to control the vehicle is on our backlog, but with no ETA.

Shibodd, 2024-05-14:
I assume steering_input is just visual, too?

aivora-beamng, 2024-05-14:
Hi, `steering_input` is a value in the range of `[-1; 1]` and it is used to actually steer the vehicles.

 https://documentation.beamng.com/modding/vehicle/vehicle_system/electrics/