BeamNG/BeamNGpy #225: cling is invalid

yuyuyu223, 2023-07-02:
```
cling: If True, the z-coordinate of the vehicle's position will be set to the ground level at the given
           position to avoid spawning the vehicle below ground or in the air.
```

Although cling=True, the cars still fly on the sky or flow in the water

[image]

I think the z-axis value does not set to the ground level.

1) How can I get the ground level of each map?
2) Is the cling a bug?

dstark481, 2023-07-02:
Hello,
There is a problem with cling, unfortunately.  We are looking into it.  You can press F11 and go into the 'world editor'.  From there, you can find the world coordinates of everything in the map.  On the left, you can see the 'scene tree' where everything is, or you can click on things which you are interested in.  When you get the z-coordinate of the ground, you can try setting the vehicle to take the position (or maybe eg 0.5m above it).
[image]

Best regards,
BeamNG Tech Team

AbdelrahmanElsaidElsawy, 2025-12-07:
Hi @yuyuyu223,  

Thanks for reporting this issue! At the moment, the *cling* feature is still listed in our backlog and hasn’t been fixed yet.

Do you have any further questions or need assistance with anything else in the meantime? You’re also welcome to join the conversation or share feedback with other developers on our [BeamNG.tech forum](https://forum.beamng.tech/).  

All the best,  
Abdulrahman Saeed  
Research Software Engineer  
BeamNG.tech Support Team