BeamNG/BeamNGpy #230: Threading with Multiple Cameras

XDynames, 2023-08-29:
I have been trying to make a setup that allows me to read from the different visual outputs of cameras without having the wait between each type or let processing time of the received output impact other readings time. I have released that directly threading calls to `.poll()` is a bad idea as the socket connection being used for all communications is the same underlying socket started by the `BeamNgPy` instance. To remedy this I start to make a different `BeamNgPy` instance per camera, per visual output. This "works" to some extend but I will usually get a few instances where `.poll()` will work initially and provide an output but then the shared memory never gets updated and I continue to read the same frame from the buffer for the runtime of the program.

I should add that this is not deterministic, seemingly randomly or maybe based on the order in which the sensors are instanced I can get any two pairs of visual outputs working and have even thought I fixed it when they all worked on rare occasions.

In this case depth and colour have continued to render bit annotation has been the same since the car spawned in:
[image]

cpap-beamng, 2023-08-30:
As a workaround, could the multishot camera help you? : https://github.com/BeamNG/BeamNGpy/blob/master/examples/multishot_camera.ipynb . 
In any case, we are investigating the issue.

aivora-beamng, 2023-10-06:
Hi, this bug should be fixed with BeamNG.tech 0.30 and BeamNGpy 1.27, you can try updating your environment.

XDynames, 2023-10-06:
I’ll let you know ASAP thank you for updating me

closed 2023-10-08 by XDynames