BeamNG/BeamNGpy #231: About lider sensor

ccc-fire, 2023-09-15:
### I found an issue. The screen in the game would go dark when the lidar sensor was attached on the car, and I  set the setting is_visualised=false. The scene brightness is not restored, which I thought was a mistake.
[image]
### When I add lider...
[image]
#### Maybe someone can help me answer that question.

AbdelrahmanElsaidElsawy, 2023-09-15:
Hello @ccc-fire 

Thank you for reporting the LiDAR sensor issue affecting the camera view, even when LiDAR visualization is off. Our team is actively working on a fix for the next release. Please feel free to share any additional feedback.

Best regards,

aivora-beamng, 2023-10-06:
Hi, this issue is fixed for BeamNG.tech 0.30 and BeamNGpy 1.27.

closed 2023-10-06 by aivora-beamng