BeamNG/BeamNGpy #232: Won't load any Scenario just main menu

devimauz, 2023-10-02:
Hey,
My problem is it don't want to load any scenario just opens the game and puts me in the main menu. 
Under Windows it works well but not on Ubuntu 20
Sometimes its give the loading screen, shows like everything loaded but nothing happens.
BeamNG.drive is 0.29.1.0.0 (BeamNG.tech)

```python
akkonut@local:~/BeamNGpy$ /home/akkonut/.pyenv/versions/3.11.4/bin/python /home/akkonut/BeamNGpy/examples/west_coast_radar.py
2023-10-03 00:45:19,607 |INFO     |beamngpy                      |Started BeamNGpy logging.
2023-10-03 00:45:19,614 |WARNING  |py.warnings                   |/home/akkonut/.pyenv/versions/3.11.4/lib/python3.11/site-packages/beamngpy/connection/connection.py:-1: ResourceWarning: unclosed <socket.socket fd=3, family=2, type=1, proto=0, laddr=('0.0.0.0', 32832)>

2023-10-03 00:45:19,614 |WARNING  |py.warnings                   |/home/akkonut/.pyenv/versions/3.11.4/lib/python3.11/site-packages/beamngpy/connection/connection.py:-1: ResourceWarning: unclosed <socket.socket fd=3, family=2, type=1, proto=0, laddr=('0.0.0.0', 32832)>

2023-10-03 00:45:19,615 |INFO     |beamngpy.BeamNGpy             |Opening BeamNGpy instance.
2023-10-03 00:45:19,615 |INFO     |beamngpy.BeamNGpy             |Started BeamNG.
2023-10-03 00:45:29,615 |INFO     |beamngpy.BeamNGpy             |Connecting to BeamNG.tech at: (localhost, 64256)
2023-10-03 00:45:30,600 |INFO     |beamngpy.BeamNGpy             |Successfully connected to BeamNG.tech.
2023-10-03 00:45:30,600 |INFO     |beamngpy.BeamNGpy             |BeamNGpy successfully connected to BeamNG.
``` 

Thanks!

aivora-beamng, 2023-10-03:
Hi, are you using Ubuntu to run BeamNGpy and BeamNG.tech, or just BeamNGpy and the simulator is running on Windows?

devimauz, 2023-10-03:
I'm using BeamNGpy and BeamNG.tech on Ubuntu.

devimauz, 2023-10-05:
I just found out gridmap is not avaiable anymore. Have to use smallgrid but gives no error for it like map not found.

[    scenario = Scenario('gridmap', 'RADAR_demo', description='Spanning the map with a RADAR sensor')](https://github.com/BeamNG/BeamNGpy/blob/82e19fb61f09d0b0347323e7e4b3187961e1d014/examples/west_coast_radar.py#L15)

I tired to run my code but sadly Shared Memory still not available under Linux?

```
import time

from beamngpy import BeamNGpy, Scenario, Vehicle, set_up_simple_logging
from beamngpy.sensors import Camera

def main():
    set_up_simple_logging()

    beamng = BeamNGpy('localhost', 64256)
    bng = beamng.open(launch=True)

    scenario = Scenario('smallgrid', 'Camera demo', description='Camera stream test')

    vehicle = Vehicle('ego_vehicle', model='etk800', license='RED', color='Red')

    scenario.add_vehicle(vehicle)
    scenario.make(bng)

    bng.settings.set_deterministic(60) # Set simulator to 60hz temporal resolution

    bng.scenario.load(scenario)
    bng.ui.hide_hud()
    bng.scenario.start()

    # NOTE: Create sensor after scenario has started.
    cam1 = Camera(
        'camera1',
        bng,
        vehicle,
        is_using_shared_memory=True,
        pos=(-5, 0, 1),
        dir=(1, 0, 0),
        field_of_view_y=70,
        is_render_annotations=False,
        is_render_depth=False,
        near_far_planes=(0.1, 1000),
        resolution=(3840, 2160),
        requested_update_time=0.05)

    t_last = 0.0
    for _ in range(10000):
        time.sleep(0.015)
        readings_data = cam1.poll()
        t = time.time()
        dt = t - t_last
        t_last = t
        print("time since last poll (s): ", dt)

    cam1.remove()
    bng.close()

if __name__ == '__main__':
    main()
```
Tired to use without it.

`beamngpy.logging.BNGValueError: This feature is not yet supported on Linux hosts.`

Attach is not supported?

Thanks!

aivora-beamng, 2023-10-06:
Hi, unfortunately, shared memory is still not supported on Linux hosts, the same holds for the `Camera` sensor. It is on the roadmap, but it goes hand-in-hand with the stabilization of the Linux version as a whole.

AbdelrahmanElsaidElsawy, 2025-12-07:
Hello @devimauz,  

Thank you for reporting this issue!  

Shared memory support for **Linux** was officially introduced in **BeamNGpy v1.30**. Since the current release is **v1.34.1**, this feature should now be functionING with some known issues, please refer to [Changlog](https://github.com/BeamNG/BeamNGpy/blob/master/CHANGELOG.rst) for details.  

Please try updating both **BeamNGpy** and **BeamNG.tech** to their latest versions to ensure full compatibility:  

```bash
pip install --upgrade beamngpy
```

After updating, try running your scenario again.  

If the issue persists with the latest versions, please share the following details so we can investigate further:  
- Your operating system and version  
- The **BeamNG.tech** build number  
- The **BeamNGpy** version 

Thanks again for bringing this up — your feedback helps us improve BeamNGpy and BeamNG.tech for everyone.  

All the best,  
Abdulrahman Saeed  
Research Software Engineer  
BeamNG.tech Support Team

devimauz, 2025-12-07:
> [Követő kép][https://github.com/notifications/beacon/ACUPEPVVE7SWQECLG65OEH34ARFCBA5CNFSM6AAAAACOJS7NWCWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTWX5ELIY.gif]

Hi,
This is very good news, I can finally test it, can I request a license again? Thank you, Gergo