BeamNG/BeamNGpy #235: How to get slip angle and cornering force ?

Rudicito, 2023-10-09:
I can't find a way to get this data. Can somebody help ?

dstark481, 2023-10-09:
Hi,
These are not quantities which we have immediately available.  Can you maybe tell us more about what it is you are trying to do with them?  Maybe there are other solutions.

Rudicito, 2023-10-10:
Thank you for responding ! I want to try to create a reinforcement learning agent that can drive a car as fast as possible on a track. I think slip angle and cornering force could be good values to input into the RL.

HighPriest, 2023-11-15:
Hi @Rudicito 
I am working on a similar task to what you are trying to achieve.
What I recommend for anyone tackling this task is to focus on information that can be obtained from real life sensors.

A good start would be rotational speed of wheels & acceleration (G forces) in at least X/Y plane. If you don't get acceleration straight from the sim, you can calculate it yourself from displacement.