BeamNG/BeamNGpy #237: Unable to launch beamng.texh on linux

aman-mehra, 2023-10-10:
Hi, I'm trying to run the example provided in the README but find that I'm unable to launch Beamng.tech on running `bng.open()`. The code fails with the following error.

```
Failed to init bngenv: Failed to init app instance frameworkError connecting to BeamNG.tech. 25 tries left.
[Errno 111] Connection refused
```

fyi - I am using a system with RHEL 7.9 and running the latest versions on Beamng.tech (0.30) and beamngpy (1.27)

Thanks!

aivora-beamng, 2023-10-11:
Hi,
does the BeamNG.tech window show at all? If not, then the process crashed and there's a bug with the Linux port which is still in the experimental phase of development.

You can try to post your issue in the support thread for BeamNG Linux port, which is located here:
https://www.beamng.com/threads/linux-port-%E2%80%93-feedback-known-issues-and-faq.86422/

aivora-beamng, 2025-01-27:
Closing for inactivity, the Linux support has improved in the past few releases, feel free to reopen if you have any new issues.