BeamNG/BeamNGpy #238: Remote connections not working as intended

Crook3dTeeth, 2023-10-21:
With the latest version of beamngpy I can't remotely connect to a beamngpy instance on another computer.  Using beamngpy version 1.26.1 works perfectly fine but 1.27 gets a WinError 10060.  I've tried disabling firewalls and all that and I've used wireshark to see that the packets are making it to the host machine but nothing is being returned or showing up in the console log.  I am also having latency issues when connecting remotely and locally with getting camera data which I can fix with is_using_shared_memory being set to true but this crashes when trying to use it remotely which is what I intend on doing.

aivora-beamng, 2023-10-23:
Hi,
for security reasons, BeamNGpy since 1.27 launches BeamNG to listen on the local network interface only by default, you can set it up to listen on all network interfaces by:

```python
beamng = BeamNGpy('localhost', 64256)
beamng.open(listen_ip='*')
```
or you can set a binding IP address (see [the docs](https://beamngpy.readthedocs.io/en/latest/beamngpy.html#beamngpy.BeamNGpy.open)) to listen on a network interface of your choice.

Shared memory cannot work with a remote host, but the user experience should be better than just crashing. We'll investigate it and put more meaningful error messages for this case.

AbdelrahmanElsaidElsawy, 2025-12-07:
> Hi, for security reasons, BeamNGpy since 1.27 launches BeamNG to listen on the local network interface only by default, you can set it up to listen on all network interfaces by:
> 
> beamng = BeamNGpy('localhost', 64256)
> beamng.open(listen_ip='*')
> 
> or you can set a binding IP address (see [the docs](https://beamngpy.readthedocs.io/en/latest/beamngpy.html#beamngpy.BeamNGpy.open)) to listen on a network interface of your choice.
> 
> Shared memory cannot work with a remote host, but the user experience should be better than just crashing. We'll investigate it and put more meaningful error messages for this case.

Hi @Crook3dTeeth. 

Thanks again for following up on this! Adam’s comment above outlines the cause and the solution for the remote connection issue in BeamNGpy 1.27.  

If everything’s now working correctly on your setup, please let us know so we can update the issue’s status. And if you’d like to discuss further configurations or performance tips, you’re very welcome to make a new issue or join the conversation on our community forum:  
**https://forum.beamng.tech/**  

All the best,  
Abdulrahman Saeed  
Research Software Engineer  
BeamNG.tech Support Team

closed 2026-01-13 by EPI-BeamNG