BeamNG/BeamNGpy #239: Flash message with only number

Guaziboi, 2023-10-27:
Im using the flowgraph to try to accomplish this. So the user drives through a trigger which should cause the message to pop up on the upper left corner. I want to say "Your speed was...". I have the flowgraph setup to save the users speed and steering angle as separate variables and they need to be displayed as messages. How can I use these variables even though messages doesn't like have just numbers be an input?

aivora-beamng, 2023-10-30:
Hi,
I think the best option is to use the "Format String" Flowgraph node to accomplish this:
[image]

It works in a similar way as the `string.format` function in Lua. 

You can download the Flowgraph example from here:
[format_string.flow.json](https://github.com/BeamNG/BeamNGpy/files/13202278/format_string.flow.json)

Guaziboi, 2023-10-30:
This worked great, thank you! Is there a way to take the data collected into the log and export it into a .csv file using python?

aivora-beamng, 2023-10-31:
You can use the "Custom Lua" node for that (we may create a CSV output node in the future, but so far it does not exist). After the Flowgraph stops, the file will be written into your userpath (`%localappdata%/BeamNG.drive/<version>`) as `flowgraph_log_[current datetime].csv`.

[image]

[format_string_csv_log.flow.json](https://github.com/BeamNG/BeamNGpy/files/13214269/format_string_csv_log.flow.json)

Guaziboi, 2023-11-08:
Perfect, thank you!!

closed 2023-11-08 by Guaziboi