BeamNG/BeamNGpy #241: After starting BeamNG.drive with BeamNGpy there is a regular drop in frames

Dreace, 2023-12-17:
After starting BeamNG.drive with BeamNGpy there is a regular drop in frames, starting it directly is fine.

The normal performance monitoring charts are shown below:
[image]
[image]

The performance monitoring charts after starting with BeamNGpy are as follows:
[image]
[image]
Performance monitoring charts show significantly higher CPU Bookkeeping times (blue raised curve)
Although the average FPS is 60+, the minimum FPS is less than 10.

Using `E:\SteamLibrary\steamapps\common\BeamNG.drive\BeamNG.drive.exe -rport 64256 -nosteam -console -tcom-listen-ip 127.0.0.1 -lua registerCoreModule('tech/techCore') -userpath C:\Users\<username>\AppData\Local\BeamNG.drive` Startup reproduces the problem, removing the ` -lua registerCoreModule('tech/techCore')` and remove `-lua registerCoreModule(') ` and the problem disappears.

It is assumed that the problem is caused by the `tech/techCore` module, but after looking at the code there is no obvious problem.

Version:
BeamNGpy: 1.27.1
BeamNG.drive: 0.31

aivora-beamng, 2023-12-18:
Hi,

1) I tried the same with BeamNGpy 1.27.1 and BeamNG.drive 0.31 and could not reproduce the performance issue. You say it happens even when BeamNGpy is not running, just when the `techCore` extension is loaded?

2) BeamNGpy 1.27.1 is not compatible with BeamNG.tech 0.31, BeamNGpy 1.28 will be released later this week. Also, we're changing the way of loading the .tech extension in BeamNGpy 1.28 and BeamNG.tech 0.31, the new command-line arguments are:
```bash
Bin64\BeamNG.tech.x64.exe -nosteam -console -lua extensions.load('tech/techCore');tech_techCore.openServer(64256)
```
The old way should still work, but is getting deprecated in the future. Can you try with these arguments and tell me if the problem still occurs?

Thank you for your detailed report.

Dreace, 2023-12-18:
Thank you for your reply.
I've tried many combinations of situations and the only regular frame dropping occurs when I start my customized scene (with GE lua script) with BeamNGpy, I'll try to find the problem in my code.

closed 2023-12-18 by Dreace